cyberman54/ESP32-Paxcounter #299: Add configurations to send data over wifi

davidex720, 2019-03-07:
I think it could be useful to have a setting and choose between sending the payload over lora or wifi.
In many countries lora has still some problems and this option could really benefit this project

cyberman54, 2019-03-07:
I will not implement this, because i am not convinced that this makes sense, since there will be a huge trade off in using the 2,4Ghz airtime for sniffing versus transmitting data.

But you're welcome to bring a pull request for this feature, i will be lucky reviewing it.

davidex720, 2019-03-07:
I see your point
This feature surely is not fast to implememet but probably scanning every minute is not strictly necessary for most of applications

cyberman54, 2019-03-07:
You can reuse existing pieces of code to implement this:
- connecting to wifi network -> see OTA.cpp
- transferring counter data to the target host -> implement additonal sendqueue, like in SPISLAVE.cpp
- cycle between sniffing/transferring -> use time-of-day (see TIMEKEEPER.cpp) or use Ticker.h to kickoff task

Downsides will be:
- loading esp32 wifi & tcp/ip stack will have (probably massive) impact on RAM, sniffing performance and power consumption
- using the wifi stack in coexistence for sniffing and transferring will habe additional impact on sniffing performance

Thus i would expect a trade-off.

cyberman54, 2019-03-07:
What you can do to decrease the tradeoff: use compiler directives to compile the code without the LoRa LMIC stack, set HAS_LORA to not defined (= comment out) in the board's hal file.

This will free up resources on Core 1 and RAM.

If you implement coexist sniffing and transferring via Wifi be aware of race conditions when accessing the wifi subsystem.  The wifi logic of paxcounter is running on Core 0.

davidex720, 2019-03-07:
this could be a good occasion to develop this point too:
https://github.com/cyberman54/ESP32-Paxcounter/issues/250

I'm on the 250

cyberman54, 2019-03-07:
Unfortunately the whole code needs some refactoring, because it was grown without a roadmap. Thus the #includes got confusing.

But the code works :-)

valentt, 2019-12-01:
If you scan wifi for 5 minutes, then you need around 5 seconds to switch from monitor mode to client mode, connect to wifi, get ip address via dhcp and send the data. You loose 5 seconds of scanning time, but in 5 minute window this is totally acceptable. For some even 5 seconds of 1 minute would be acceptable.

cyberman54, 2019-12-02:
@valentt It's not necessary to manage a wifi connection while sniffing MACs. Sniffing operates in the background, even while a wifi connection is established and in use.

But of course it is necessary to stop wifi channel rotation while a wifi connection is established. This could easyly be implemented using ESP32 wifi events, calling `xTimerStop(WifiChanTimer, 0);` before wifi connection is established, and `xTimerStart(WifiChanTimer, 0);` after it is released.

Stopping wifi channel rotation will, of course, reduce the MAC discover rate, theoretically by ~90% (1/13).

Besides this, an established wifi connection could perhaps reduce the sniffer performance, when packets get lost. This is an assumption, i never tested this.

You're welcome to create a PR for the wifi data transfer feature.

cyberman54, 2020-05-20:
@davidex720 we have some news here. In development branch there is an enhancement with mqtt client, sending pax telemetrics data over ethernet interface.

holgerkoch, 2021-03-19:
Hi,
i have a question for sending the pax telemetrics over WiFi. 
In paxcounter.conf is a config item:
#define MQTT_ETHERNET 0 // select PHY: set 0 for Wifi, 1 for ethernet
and i interpret it that the paxcounter connect to wifi with the credentials from the ota.conf settings and will be send the telemetry to a mqtt server.
Unfortunatelly, he doesn't connect to wifi. If i activate the bootmenu, he connect and start the webserver successful.

Should that work? 

Best regards

cyberman54, 2021-03-19:
as stated above: wifi connect during runtime is not yet implemented. MQTT only works via Ethernet PHY.

holgerkoch, 2021-03-19:
@cyberman54,
thank you very much for your quick response.
So i will try to use lora.
Maybe it is a good idea to add a comment, that the config item  

#define MQTT_ETHERNET 0 // select PHY: set 0 for Wifi, 1 for ethernet

will not work for wifi, because it is not implemented now.

Best regards, Holger

cyberman54, 2021-03-19:
Comment added. Thanks for the hint.

raomin, 2023-01-29:
Hello, I would also need this feature. Before considering a PR, I would like to know if the approach described by @cyberman54 [here](https://github.com/cyberman54/ESP32-Paxcounter/issues/299#issuecomment-560292232) is still the right one.

cyberman54, 2023-01-29:
@raomin Yes, it is. An approach would be like this:
- if send cycle is due, stop wifi sniffing, e.g. by calling `set_wifiscan 0` from _rcommand.cpp_
- start wifi stack in station mode
- connect to wifi ap and send data (e.g. using `start_ota_update(`) from _ota.cpp_ as boiler template)
- stop wifi stack
- restart wifi sniffing,, e.g. by calling `set_wifiscan 1` from _rcommand.cpp_

raomin, 2023-01-29:
I see I cannot/should not call set_wifiscan but go through 
``c
void rcommand(const uint8_t *cmd, const size_t cmdlength)
``
There are no implementation of this function call for a specific parameter and the implementation is not obvious at first glance...
Can you save me some time and explain how I should send one call for `{0x17, set_wifiscan, 1}` to `0`?
Thanks

cyberman54, 2023-01-29:
set_wifiscan(0) stops wifi sniffing, (1) starts

raomin, 2023-01-29:
This I saw (and did). It's just that only `rcommand()` was declared in `rcommand.h` as this looks like the entry point of commands that are redirected to other private functions eg `set_wifiscan`.
But if it's ok to you to add a declaration of `set_wifiscan` in   `rcommand.h` it's fine with me! :)

raomin, 2023-01-29:
So I started working on this and I guess made the required modifications.
You can have a look at my fork https://github.com/raomin/ESP32-Paxcounter

I have implemented the following step in mqttclient: 
1. stop wifiscan
2. connect on wifi
3. connect to mqtt to send whatever needs to be sent
4. I disconnect wifi
5. and then start wifiscan.

Thing is, when I start wifiscan again, I get an out of mem crash... ESP_ERR_NO_MEM
Any luck this can be fixed?

```log
00:12:27.135 > ets Jun  8 2016 00:22:57
00:12:27.135 >
00:12:27.135 > rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:12:27.135 > configsip: 0, SPIWP:0xee
00:12:27.135 > clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:12:27.135 > mode:DIO, clock div:2
00:12:27.135 > load:0x3fff0030,len:1184
00:12:27.135 > load:0x40078000,len:13104
00:12:27.135 > load:0x40080400,len:3036
00:12:27.135 > entry 0x400805e4
00:12:27.135 > I (429) src/configmanager.cpp: Loading device configuration from NVRAM...
00:12:27.135 > I (431) src/configmanager.cpp: Runtime configuration v3.4.62 loaded
00:12:27.135 > D (432) src/reset.cpp: Timezone set to CET-1CEST,M3.4.0/2,M10.4.0/3
00:12:27.135 > I (437) src/main.cpp: Starting paxcounter_23da1aa7 v3.4.62 (runmode=0 / restarts=0)
00:12:27.135 > I (444) src/main.cpp: code build date: 1675033412
00:12:27.135 > I (449) src/main.cpp: This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
00:12:27.135 > I (458) src/main.cpp: Internal Total heap 205108, internal Free Heap 176516
00:12:27.135 > I (465) src/main.cpp: ChipRevision 1, Cpu Freq 240, SDK Version v4.4.3
00:12:27.135 > I (471) src/main.cpp: Flash Size 4194304, Flash Speed 40000000
00:12:27.135 > I (477) src/main.cpp: Wifi/BT software coexist version 1.2.0
00:12:27.135 > I (482) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-i2c.c: Initialising I2C Master: sda=21 scl=22 freq=100000
00:12:27.135 > I (497) src/i2c.cpp: Starting I2C bus scan...
00:12:27.135 > W (500) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp: Bus already started in Master Mode.
00:12:27.135 > I (630) src/i2c.cpp: 0 I2C device(s) found
00:12:27.135 > I (3158) src/power.cpp: ADC characterization based on reference voltage stored in eFuse
00:12:27.135 > I (3162) src/main.cpp: Starting libpax...
00:12:27.135 > I (3162) src/main.cpp: WIFISCAN: on
00:12:27.135 > I (3162) src/main.cpp: BLESCAN: on
00:12:27.135 > I (3165) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Initializing bluetooth scanner ...
00:12:27.135 > I (3891) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: BLE Scanning started
00:12:27.135 > I (4092) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Bluetooth scanner started
00:12:27.135 > I (4101) src/main.cpp: Starting rcommand interpreter...
00:12:27.135 > I (4101) src/rcommand.cpp: Rcommand send queue created, size 55 Bytes
00:12:27.135 > I (4102) src/mqttclient.cpp: MQTT send queue created, size 510 Bytes
00:12:27.135 > I (4107) src/mqttclient.cpp: Starting MQTTloop...
00:12:27.135 > I (4112) src/rcommand.cpp: Remote command: set WIFI scanner to off
00:12:27.135 > I (4118) src/main.cpp: Starting Interrupt Handler...
00:12:27.135 > I (4120) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Shutting down bluetooth scanner ...
00:12:27.135 > I (4131) src/main.cpp: Starting Timers...
00:12:27.135 > I (4134) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Bluetooth scanner stopped
00:12:27.135 > I (4142) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Initializing bluetooth scanner ...
00:12:27.135 > I (4185) src/button.cpp: Starting button Controller...
00:12:27.135 > I (4186) src/main.cpp: Features: DISP BATT OTA MQTT PACKED BTN_PU
00:12:27.135 > I (4788) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: BLE Scanning started
00:12:27.135 > I (4989) .pio/libdeps/usb/libpax/lib/libpax/blescan.cpp: Bluetooth scanner started
00:12:27.135 > D (5014) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: Arduino Event: 0 - WIFI_READY
00:12:27.135 > D (5020) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: Arduino Event: 2 - STA_START
00:12:27.135 > D (5086) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: Arduino Event: 4 - STA_CONNECTED
00:12:27.135 > D (5520) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: Arduino Event: 7 - STA_GOT_IP
00:12:27.135 > D (5522) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: STA IP: 192.168.1.54, MASK: 255.255.255.0, GW: 192.168.1.1
00:12:27.135 > I (5542) src/wifiManager.cpp: Trying to connect to romiot, attempt 1 of 5
00:12:27.135 > D (5543) src/mqttclient.cpp: MQTT client reconnecting...
00:12:45.779 > I (25548) src/mqttclient.cpp: MQTT name is paxcounter_23da1aa7
00:12:45.781 > I (25549) src/mqttclient.cpp: Attempting to connect to 192.168.1.46 [192.168.1.46]
00:12:45.797 > I (25573) src/mqttclient.cpp: MQTT server connected, subscribing...
00:12:45.893 > I (25667) src/mqttclient.cpp: MQTT topic subscribed
[... 8 lines cut ...]
00:12:55.931 > W (35676) C:/Users/bourgur/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp: Reason: 8 - ASSOC_LEAVE
[... cut ...]
00:12:57.471 >
00:12:57.471 > abort() was called at PC 0x40097127 on core 1
00:12:57.471 >
00:12:57.471 >
00:12:57.471 > Backtrace: 0x40084525:0x3ffff110 0x40097131:0x3ffff130 0x4009dced:0x3ffff150 0x40097127:0x3ffff1d0 0x400d8b05:0x3ffff1f0 0x400d8916:0x3ffff2e0 0x400d46b4:0x3ffff300 0x400d7a71:0x3ffff320 0x400d8685:0x3ffff360 0x400d4f9a:0x3ffff390
00:12:57.481 >
00:12:57.481 >
00:12:57.481 >
00:12:57.481 >
00:12:57.481 > ELF file SHA256: 0e4ba2be5bb58df2
00:12:57.481 >
00:12:57.867 > Rebooting...
```

cyberman54, 2023-01-30:
yes, use the private function and declare it as public, that should work.

cyberman54, 2023-01-30:
don't touch ota.cpp and ota.h, both are not needed here. 
Try to unload wifi driver from RAM, after you finished you comunication, there is a ESP-IDF API call for this.
Try using a board with additional PSRAM.

raomin, 2023-01-30:
See my fork here.
https://github.com/raomin/ESP32-Paxcounter
I though this was a good idea to implement a WifiManager that would centralize the Wifi switches for OTA and MQTT. But if you prefer, I just duplicate code.

On your advice, I switched to an esp32s3. That's so much better. 
The code works fine now on an esp32s3. I guess the ble stack just takes too much on an esp32.
No more short of memory or weird reboots. Thanks!

Now the MQTT message is a byte version encoded in base64.
I guess a json message would be more appropriate. 
Are you OK with defining a MQTT_ENCODER configuration item with 0 for base64 and 1 for JSON?

cyberman54, 2023-01-30:
The Wifimanager in arduino-esp32 was not reliable over a long time, which is fatal for OTA. Didn't catch up with latest versions. If it works reliable now it could make sense to centralize wifi code, using same functions for OTA and MQTT.

ESP32-S3 has same RAM as base ESP32. Perhaps you switched to an S3 board which sports SPI PSRAM?

I think it would be possible to have code with less RAM footprint, if you don't use the Wifimanager, and unload wifi stack from RAM, after using it.

There is no BLE stack in paxcounter, neither in libpax. The code is using the low level VHCI API of ESP32 to control BLE, thus does not have a significant RAM footprint.

Feel free to change MQTT encoding/protocol, but please provide an appropriate update/addendum of documentation. Thanks.

raomin, 2023-01-31:
- Wifi
I was not refering to the arduino project WifiManager but merely to the [`wifiManager.cpp`](https://github.com/raomin/ESP32-Paxcounter/blob/master/src/wifiManager.cpp) code I've put centralizing the wifi connections and disconnections from [`ota.cpp`](https://github.com/raomin/ESP32-Paxcounter/blob/master/src/ota.cpp) and [`mqttclient.cpp`](https://github.com/raomin/ESP32-Paxcounter/blob/master/src/mqttclient.cpp)

Let me know if you prefer I leave ota.cpp untouched, I would remove `wifiManager.cpp` then.

- RAM
Yes, I'm using an esp32-s3, with psram :) But I'll also unload wifi stack. I guess calling `esp_wifi_stop()` then `esp_wifi_deinit()` should do the job... 

- MQTT/JSON
Most MQTT implementation I've worked with do not include message handler (eg nodered) that would unpack binary. It still remains within a tcp packet ;)
Anyway, that remains an option in my implementation:
I have the json encoding working if you want to [have a look.](https://github.com/raomin/ESP32-Paxcounter/blob/ccc876b6251f6750ea0074a5d5a1af7217c5855d/src/mqttclient.cpp#L111) 
```
paxout paxcounter_9131294e
paxout/1 {'total':1, 'ble':0, 'wifi':1}
```

Now one caveat of on demand wifi/mqtt connection is that it's unlikely that we receive a rcommand mqtt message at the same time on the subscribed topic. Unless the message is send in _retain_ but then it would be read each time. 

I'll put a a quick doc update to clarify MQTT use and this last point. Also a bit of code cleanup and it should be ready for PR.

raomin, 2023-02-01:
So, I'm trying to get something working on esp32 (no psram).
Thing is, I cannot even initialize the interrupt: no more memory.

Here are my findings on memory consumption:
[image]

Sure, I could remove display, and it would work, or just stick to psram... But is it hopeless to slim up libpax memory consumption?

cyberman54, 2023-02-02:
That's strange. Libpax assigns 2KB of DRAM as bitmap for storing identifiers. It does not load a wifi stack, neither a bluetooth stack. It does not include any external libraries. How did you create the memory map?

raomin, 2023-02-02:
ESP.getFreeHeap() at every step of the setup.
Going deeper in this rabbit hole, that's the `wifi_sniffer_init` that consumes most of the 98kB. Nothing much we can do here I think...

cyberman54, 2023-02-03:
Thanks for investigating this. Wifi_sniffer_init in libpax calls esp32 idf functions, i wouldn't expect that these cause such big impact on RAM. Maybe it's the promiscous mode callback which is called with every wifi packet arriving. How many wifi devices are counted in your test environment by paxcounter?

cyberman54, 2023-02-03:
Libpax is using static memory. It's assigned to a bitmap array, at this point in code:
https://github.com/dbSuS/libpax/blob/master/lib/libpax/libpax.cpp#L33
The calculation of array size should result to 2KB. But maybe it isn't for some reason?

cyberman54, 2023-02-03:
@raomin 
Could you please try to switch off wifi sniffer, only use BLE sniffer, then again check free ram?
1. Set WIFICOUNTER 0 and BLECOUNTER 1 in paxcounter.conf
2. Delete NVRAM by pio run -t erase 
3. Rebuild and reflash software
4. Run and inspect RAM
Thanks.

raomin, 2023-02-03:
> Libpax is using static memory. It's assigned to a bitmap array, at this point in code: https://github.com/dbSuS/libpax/blob/master/lib/libpax/libpax.cpp#L33 The calculation of array size should result to 2KB. But maybe it isn't for some reason?

Just to clarify what I'm doing for memory measurement, at every step of `setup()` I'm printing `ESP.getFreeHeap()` and deduce what is consumed by the function just called. So static memory is consumed before I get anything printed.
Also, because of the use of multithreading, there could be a side impact on the memory that I wrongly assign to `Wifi_sniffer_init`...
Let's see with WIFICOUNTER 0...

cyberman54, 2023-02-03:
Maybe a false positive?

`ESP.getFreeHeap()` boils down to RTOS `heap_caps_get_free_size(MALLOC_CAP_INTERNAL)`. 
This shows only [a part of the whole memory](https://github.com/espressif/esp-idf/blob/master/components/heap/include/esp_heap_caps.h#L20-L37).

raomin, 2023-02-03:
> Maybe a false positive?
> 
Not really, after my init on a tdisplay, with DISP, BLE, WIFI and MQTT, `ESP.getFreeHeap()` return less than 4Kb and it crashes when it tries to initiate the 4Kb of irqHandlers for the state machine. If I disable DISP for ex, getFreeHeap is higher and I pass the irqHandlers. So that's pretty representative of what memory is left. 

I tried with WIFICOUNTER to 0 and free heap is now 74kb more than when WIFICOUNTER is to 1.

cyberman54, 2023-02-03:
Hm, what may be causing libpax to draw 74kb for wifi sniffing?

cyberman54, 2023-02-04:
Did some investigation. The esp32 wifi driver consumes around 71 KB of RAM. It's memory footprint can be lowered by tuning some buffer sizes, see https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#why-buffer-configuration-is-important

I found that this can free up 10 - 15 KB of RAM.

raomin, 2023-02-04:
So I guess we should see with `static void useStaticBuffers(bool bufferMode);` to `true`...
I'll have a look

cyberman54, 2023-02-05:
Paxcounter generally does not use Wifi class, except for OTA, but it then reboots before to free RAM, before running ota.cpp.

raomin, 2023-02-05:
But then, is the sniffer actually using these buffers?

cyberman54, 2023-02-05:
Not sure. I did not find a documentation source yet which explains the sniffer mode of esp32 api.

cyberman54, 2023-02-05:
@raomin You should not use Arduino WIFI class function calls from within paxcounter code, because paxcounter delegates control of wifi to the libpax library. Libpax is based on esp32 api wifi function calls, not on Arduino WIFI class.

Only exception in paxcounter is ota.cpp which is using Arduino WIFI class to establish a tcp/ip connect over wifi, but this happenes i a special mode, after the module is rebooted for, thus libpax is not active while ota.cpp is running.

Regarding memory buffers for wifi usage, what Arduino WIFI class does is setting the RX/TX buffer values to it's own configuration, found [here](!WiFiGenericClass::useStaticBuffers). In libpax code buffer adjustments would need to happen  [here](https://github.com/dbSuS/libpax/blob/master/lib/libpax/wifiscan.cpp#L93). Libpax lib does not expose this to it's api, so there won't be a straight forward way to modify wifi buffers from within paxcounter code. But you could fork libpax and modify it for experiments.

raomin, 2023-02-05:
Thanks for the clarification. That's more or less what I did already with `set_wifiscan(0)` which calls `libpax_counter_stop()`.
It's just that I'm troubled with the few memory we have left which is not enough for the wifi stack, the MQTT stack, the display and the sniffing of BLE and WIFI. That leaves us with either unloading everything (which I find challenging in a nice way) or... using spram :)

cyberman54, 2023-03-05:
A preliminary version of this feature can be found in branch mqtt-over-wifi

cyberman54, 2025-09-13:
Another version which implements this feature can now be found in branch development-mqtt-over-wifi