cyberman54/ESP32-Paxcounter #304: Build for ttgov21new with HAS_BME fails

gregorwolf, 2019-03-07:
I've added

```
#define HAS_BME GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL
#define BME_ADDR BME680_I2C_ADDR_PRIMARY // connect SDIO of BME680 to GND
```

to **ttgov21new.h** and tried to build the project. But that results in the following errors:

> .pioenvs/ttgov21new/lib214/libBosch-BSEC.a(bsec.cpp.o):(.literal._ZN4Bsec11beginCommonEv+0x0):  undefined reference to 'bsec_init'
.pioenvs/ttgov21new/lib214/libBosch-BSEC.a(bsec.cpp.o):(.literal._ZN4Bsec11beginCommonEv+0x4): undefined reference to 'bsec_get_version'
.pioenvs/ttgov21new/lib214/libBosch-BSEC.a(bsec.cpp.o):(.literal._ZN4Bsec18updateSubscriptionEP21bsec_virtual_sensor_thf+0x0): undefined reference to 'bsec_update_subscription'
...

The build with `env_default = generic` runs through without any issues.

gregorwolf, 2019-03-07:
And now solved it on my own. Just checked the settings for **[env:ttgov21new]** in **platformio.ini** and replaced  **build_flags_basic** with **build_flags_all**.

closed 2019-03-07 by gregorwolf

cyberman54, 2019-03-08:
Can you describe somewhere that build flags must be set according to hardware and make pr? thx

gregorwolf, 2019-03-08:
I've just used build_flags_all instead of build_flags_basic and had no issues. Should I put this in a pr?

cyberman54, 2019-03-08:
No. It is intentional to not include BME libs in `build_flags_all` because the librabry costs lots of memory.
It should be mentioned in the readme that modifying device configurations can need edits in platformio.ini.