cyberman54/ESP32-Paxcounter #306: Should BSEC library support BME280?

gregorwolf, 2019-03-08:
Yesterday I've made the ttgov21new compile with HAS_BME to activate the BSEC library. Unfortunately my BME280 wasn't detected. Should it work?

cyberman54, 2019-03-08:
No. The software is for BME680. But could sure be tailored for BME280, since the Bosch BSEC library is compatible with the whole family.

gregorwolf, 2019-03-08:
Because I where successful yesterday to get my BME280 running on my  ttgov21new using the [Adafruit_BME280_Library](https://github.com/adafruit/Adafruit_BME280_Library) inside Paxcounter. Should I contribute that?

cyberman54, 2019-03-08:
Please push a new branch for this, i will then have a look at it.
I remember there was a certain reason why i did not use the arduino library.

cyberman54, 2019-03-13:
BME280 added with PR #309

closed 2019-03-13 by cyberman54