cyberman54/ESP32-Paxcounter #311: Display not working on Heltec V2

muhackl, 2019-03-13:
Hello there,
I got the Heltec Lora Wifi V2 (https://www.amazon.de/gp/product/B078M74NNN/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1)
When using heltecv2 (in platformio.ini) and pushing the code to the device, the display stays blank. 
The display was working propery with the default firmware on the board.
I checked the pinouts - you even can see them on the amazon page: It's 4 for OLED_SDA, 15 for OLED_SCL and 16 for OLED_RST - as also stated here: http://www.heltec.cn/download/WIFI_LoRa_32_Diagram.pdf

So in the code heltecv2.h, everything seems fine.
One LED on the board is flashing repeatedly - I was not able to check, if it sends data as I have no LORA WAN nearby.
But even though - display is not working. How can I debug it? What might be the reason?
I'm using Microsoft Visual Studio code on Win 10 to push the firmware to the board. After compiling and after transmitting, VS is telling me, everything is fine (success).
Regards
Muhackl

cyberman54, 2019-03-13:
Please try the v1 code ob the board, that is heltec.h instead heltecv2.h, and post results here.

cyberman54, 2019-03-13:
I think the pinouts where swapped for V1 and V2 version.
I corrected both hal files and tested on a V1 version, it works with the new settings
Please reload software, reflash and try again.

muhackl, 2019-03-13:
Tested it by just activating env_default = heltec in platformio.ini and deactivating env_default = heltecv2.
Compiling anf flashing --> totally same: Blank display, LED flashing constantly. 
How can I get sure which pinouts are used?

cyberman54, 2019-03-13:
Ask Heltec. There are various erraneuos pinouts on the internet. As far as i see ALL Heltec version use 4/15 for i2c of Display.

muhackl, 2019-03-13:
I took a look at the output of the compiler. There are - despite of telling me success at the end - yellow highlighted messages: 
.piolibdeps\MCCI LoRaWAN LMIC library_ID5774\src\hal\getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
.piolibdeps\MCCI LoRaWAN LMIC library_ID5774\src\hal\getpinmap_thisboard.cpp:51:72: note: #pragma message: Board not supported -- use an explicit pinmap
#pragma message("Board not supported -- use an explicit pinmap")
^

cyberman54, 2019-03-13:
This warning can be ignored, has no impact.
I checked the 4/15 setting now on both Heltec v1 + v2, both work. Make sure you build your code libraries fresh by pressing clean in platformio before rebuild and upload.

muhackl, 2019-03-13:
Ok about the warning. 
I used your latest commit e38e79f. I pressed clean. Built again for V1 and flashed. Blank display. Cleaned, built again for V2 and flashed. Blank display.
As I am new to working with Arduino/ESP32 - perhaps I'm doing something wrong? I followed the instructions in the Heise article step by step. I checked, if the display is working by starting the board when I got it - before I flashed with paxcounter. It worked. 
My feeling is telling me, it's something totally minor I overlook. What else can I check?

cyberman54, 2019-03-13:
It sounds like your compiler did not get the changes in the .h file. This sometimes happenes with platformio. Make sure that you
0) saved all edits
1) power off board (plug off USB) and re-power
2) select appropriate board file in `platformio.ini`
3) run platformio clean (press dash bin in task bar)
4) build & flash

I tested the code on both Heltec v1 and v2 today, it runs.

muhackl, 2019-03-14:
Thanks for this hint - I will try this later. But let me describe, what I whitnessed today. 
I did not change anything after my last post. No reflash. Nothing.
I took the paxcounter to work today to be sure to get best quality TTN lora connection and a lot of devices around.
I switched it on - same thing as yesterday. LED flashing. Blank screen.
But obviously it connected to TTN and he device registered itself. Great! 
But: It also did not detect or report any devices nearby (it's an office building - just in a diameter of 20 meters there have to be 15 devices detectable).
So I left it alone on the window bench for several hours. When I got back, the display was on (!) an showing the pax count. It also transferred some data of recognized devices (I guess) to TTN. The connection to Tago IO I use for building a dashboard.obviously did not work - no data shown there.
What happened? Why did it start to work after several hours being left alone?

cyberman54, 2019-03-14:
Sorry, i can't help you here. Effect is too weird to give clear diagnostics by remote. You will have to figure out more details on yourself to get this debugged. I would recommend to use a second device for this, to compare results.

I did re-test the current software on both a Heltec v1 and v2 board yesterday, with no issue.

In general, display issues could be affected by

- settings in HAS_DISPLAY
- pin mapping of i2c interface
- NVRAM storage of display on/off mode
- wiring of i2c bus, i.e. if additional devices share same i2c bus
- power

cyberman54, 2019-03-14:
When the display shows joining, the device has not joined TTN network. It is impossible that data is transferred while device remains in this state.

But device could have joined before,  then made reset for some reason (did you check power?) and then did not rejoin.

caskerra, 2019-03-15:
Hi, I have three V2 Boards from Heltec and each behave a little distinct. - In any case 4/15 pinning is for the i2c. 
For one board I have to connect 21 (Vext control) to GND - instantly hardwired - Then this board comes up with its screen everytime. It is the only board of the three that need this additional connect, look at the picture.
[image]

cyberman54, 2019-03-15:
GPIO21 is heltec v2, the heltec does not have low power switching logic. GPIO21 can be controlled by using an option in the board's hal file, look in heltecv2.h in development branch.

closed 2019-03-15 by cyberman54

muhackl, 2019-03-16:
> Hi, I have three V2 Boards from Heltec and each behave a little distinct. - In any case 4/15 pinning is for the i2c.
> For one board I have to connect 21 (Vext control) to GND - instantly hardwired - [...]

This worked for me. Thanks!