cyberman54/ESP32-Paxcounter #314: Connecting to LoRaServer.io when the APPEUI is not available

proffalken, 2019-03-15:
Hi Folks,

I have a TTGO TBeam running PaxCounter which is working perfectly with TTN, however I want to switch it over to my private LoRaServer based on loraserver.io

I've updated the DEVEUI and the APPKEY, however the [loraserver docs](https://www.loraserver.io/lora-app-server/use/devices/) say that the APPEUI should be set to `0000000000000000`

Once I update these settings, I see a JOIN being sent to the server, followed by a JOIN ACCEPT being returned, but this is then followed in the logs with `loraserver1 loraserver[20729]: time="2019-03-15T07:04:35Z" level=error msg="processing rx packet error: get device-session error: device-session does not exist or invalid fcnt or mic"`

The device then attempts to re-join the network, resulting in the same cycle.

Does anyone have the TTGO TBEAM (or Paxcounter in general!) working with loraserver.io?

cyberman54, 2019-03-15:
invalid MIC could point to a wrong Netwks key?

proffalken, 2019-03-15:
Yeah, I thought that, but I've cut&pasted in MSB format from the LoRaServer console so I'm 99% confident that it's set correctly.

Netwks key is the same as the APPKEY, correct?

cyberman54, 2019-03-15:
Maybe you better ask in this forum:

 https://forum.loraserver.io/t/otaa-root-keys-nwkkey-from-where/1456/2

Since this is no paxcounter related issue, i close this here, but you're welcome to keep us posted here on your efforts and solution to get the device running with a loraserver.

closed 2019-03-15 by cyberman54