cyberman54/ESP32-Paxcounter #327: Bluetooth funktioniert nicht !?

tebby69, 2019-03-25:
Moin!

I have problems by counting Bluetooth-Devices. I set BLECOUNTER to 1. But my Paxcounter (T3_V1.6) shows only "BLTH: off".
Do I have to change something else?

__________________my paxcounter.conf______________ 

#define PRODUCTNAME                     "PAXCNT"

// Verbose enables serial output
#define VERBOSE                         1       // set to 0 to silence the device, for mute use build option

// Payload send cycle and encoding
#define SENDCYCLE                       30      // payload send cycle [seconds/2], 0 .. 255
#define PAYLOAD_ENCODER                 2       // payload encoder: 1=Plain, 2=Packed, 3=Cayenne LPP dynamic, 4=Cayenne LPP packed

// Set this to include BLE counting and vendor filter functions
#define VENDORFILTER                    1       // set to 0 if you want to count things, not people
#define BLECOUNTER                      1       // set it to 1 if you want to use BLE count, at expense of memory

// BLE scan parameters
#define BLESCANTIME                     0       // [seconds] scan duration, 0 means infinite [default], see note below
#define BLESCANWINDOW                   80      // [milliseconds] scan window, see below, 3 .. 10240, default 80ms
#define BLESCANINTERVAL                 80      // [illiseconds] scan interval, see below, 3 .. 10240, default 80ms = 100% duty cycle

cyberman54, 2019-03-25:
You need to enable the bluetooth function, not only to include the feature in the code.
See here:
https://github.com/cyberman54/ESP32-Paxcounter/issues/272

closed 2019-03-25 by cyberman54

tebby69, 2019-03-25:
Okay, this link was helpfull!
I modified the default setting “blescan” in configmanager.cpp an changed the version number in platformio.ini.
Now the display shows BLTH: 0 - but the counter should see some devices in the nearest surroundings!

cyberman54, 2019-03-25:
Depends on your devices. Paxcounter filters certain BT devices, which are not smartphones. You may compile without feature VENDORFILTER, this should increase number of seen devices. Rest depends on your antenna. Many china boards have crap antennas.