cyberman54/ESP32-Paxcounter #331: Running the program without Lora

PAk-CatchFire, 2019-03-31:
Hi. 
I am trying to run Paxcounter on a simple ESP32 board, to do that I am editing generic.h file to disable the defines of HAS_LORA, HAS_BME, HAS_SENSORS, HAS_DISPLAY, HAS_GPS, HAS_RTC, HAS_DCF77 and BOARD_HAS_PSRAM.
I also disabled TIME_SYNC_INTERVAL on paxcounter.conf

However I am seeing on the board:

`...
I (3381) wifi: set country: cc=EU<0> schan=1 nchan=13 policy=1

I (3457) wifi: ic_enable_sniffer
[I][main.cpp:360] setup(): Starting Interrupt Handler...
[I][main.cpp:390] setup(): Starting Timers...
[I][main.cpp:417] setup(): Starting Timekeeper...
[I][timekeeper.cpp:76] timepulse_init(): Timepulse: external (GPS)
abort() was called at PC 0x4015149f on core 0

Backtrace: 0x40094878:0x3ffb0c40 0x40094aa9:0x3ffb0c60 0x4015149f:0x3ffb0c80 0x401514e6:0x3ffb0ca0 0x401508cb:0x3ffb0cc0 0x400d466e:0x3ffb0ce0 0x400d478f:0x3ffb0d10 0x40081293:0x3ffb0d50 0x40163b7a:0x3ffb0d70 0x400909b1:0x3ffb0d90 0x40096e7d:0x3ffb0dd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
[I][main.cpp:131] setup(): Starting PAXCNT v1.7.41
[I][main.cpp:144] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
[I][main.cpp:146] setup(): Internal Total heap 259368, internal Free Heap 233656
[I][main.cpp:152] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.3-beta1-179-ge931fe9f5-dirty
[I][main.cpp:154] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:155] setup(): Wifi/BT software coexist version 1.1.3
[I][configmanager.cpp:185] loadConfig(): Reading settings from NVS
....`

Am I doing anything wrong?
Thank you

cyberman54, 2019-03-31:
You must undefine PSRAM if your board does not have sich.

Use lolinlite.h board file, this also has no LoRa.

cyberman54, 2019-03-31:
Checked lolinlite.h -> compiles with no LORA without error 
-> fixed.

closed 2019-03-31 by cyberman54

PAk-CatchFire, 2019-03-31:
Thank you for your answer.
Unfotunatelly, if I use lolin32lite I get the following error messages:

```
Compiling .pioenvs\lolin32lite\src\wifiscan.cpp.o
src\timesync.cpp: In function 'void process_timesync_req(void*)':
src\timesync.cpp:71:11: error: 'LMIC' was not declared in this scope
while (!LMIC.devaddr) {
^
src\timesync.cpp:108:24: error: 'LMIC_sendAlive' was not declared in this scope
LMIC_sendAlive();
^
src\timesync.cpp:125:42: error: 'os_getTime' was not declared in this scope
milliseconds(osticks2ms(os_getTime())) + milliseconds(TIME_SYNC_FIXUP);
^
src\timesync.cpp:125:43: error: 'osticks2ms' was not declared in this scope
milliseconds(osticks2ms(os_getTime())) + milliseconds(TIME_SYNC_FIXUP);
^
*** [.pioenvs\lolin32lite\src\timesync.cpp.o] Error 1
```
I need to comment out on paxcounter.conf the following defines:

```
//#define TIME_SYNC_COMPILEDATE           0       // set to 1 to use compile date to initialize RTC after power outage [default = 0]
//#define TIME_SYNC_LORAWAN               0       // set to 1 to use LORA network as time source, 0 means off [default = 0]
//#define TIME_SYNC_TIMESERVER            0       // set to 1 to use LORA timeserver as time source, 0 means off [default = 0]
```

and I get this output:

```
I (764) wifi: ic_enable_sniffer
[I][main.cpp:360] setup(): Starting Interrupt Handler...
[I][main.cpp:390] setup(): Starting Timers...
[I][senddata.cpp:70] sendCounter(): Counter cleared
[I][senddata.cpp:70] sendCounter(): Counter cleared
```

But I do not see any data.

cyberman54, 2019-04-01:
If you do not have LoRa,
- you must disable the LoRa based options for time synching (but `TIME_SYNC_COMPILEDATE` works)
- you cannot send any data (but using an SPI interface works, see `spislave.cpp`)