cyberman54/ESP32-Paxcounter #341: Some Smartphones not Detectable?

Zufallsfuchs, 2019-04-05:
We've experimented in an environment with nearly no WIFI disturbances and put our Smartphone very close to our PAXCounter (ttgov21new) but it didn't cause the PAX number nor the WIFI number to increase. 
Our Smartphone is from Xiaomi with the vendor address 0xA45046. The address can be found in "vendor_array.h"

Any ideas why it isn't counted for?

cyberman54, 2019-04-05:
Maybe your smartphone uses MAC adress randomization, including the OUI part of the MAC, and it is then filtered by the vendorfilter (compile the software without vendor filter to suppress this).

Maybe your smartphone does not send probe requests, for some reason.

There some verbose debug options in `macsniff.cpp`. Use highest debug level setting in `platformio.ini` and check in console messages if the MAC is seen and listed as counted or dropped.

Check the functionality of the RF part of your ESP32 board by testing with different Wifi devices, e.g. move to an area with many devices ("go out on the street to check").

cyberman54, 2019-04-05:
And: It can take some time to detect a device. Because the ESP32 chip has only one radio built in, it can only sniff on a single wifi channel at a time. In Europe 13 channels for 2,4 GHz wifi are used. Thus, paxcounter software makes time divison multiplexing the radio by cycling it over all 13 channels. Default setting is 2 channels per seconds. The time until a device is detected depends on how often a device sends probe request. You can change length and frequency of cycles in `paxcounter.ini` .

Besides this, there will be a certain rate of packet loss, esp. in environments with high wifi density, due to the limited speed of ESP32 running the paxcounter code. The code was optimized to minimize packet loss by running the wifi sniffing code exclusively on CPU core0, while all other tasks (LoRa, display, interrupts, etc.) run on core1.

And, of course, it could be a still undetected bug in the software. Since this is open source you're welcome to make pull request if you find the bug.

ajpaezm, 2019-10-28:
@Zufallsfuchs @cyberman54 does any of you know how to converte a MAC address to an encoded version of them like the ones in the vendor_array.h file? 

Suppose I have something like this: a0 b0 3c 28 3a 85

How can I change it to a format that looks like this: 0x123456?

cyberman54, 2019-10-28:
vendor_array.h consist of the vendour OUI part of MAC addresses, these are the 3 MSB octets.

`a0 b0 3c 28 3a 85` -> `0xa0b03c`

ajpaezm, 2019-11-16:
@cyberman54 btw, I'm trying to pick up some MACs using BLE sniffing, but I'm not getting anything (only an Apple TV I think, I know because of the MAC ID). I tried setting the Vendor Filter to ON and also tested it with it turned off, no changes. I updated the vendor array using with the MACs I wanted to read, with these: 

0xb0e03c, 0x000788 (Alcatel and BQ AQURIUS), updating the number of entries inside vendor_array.h as well of course. 

Also, when I turned off the BLE counter and the Vendor Filter to only read WiFi, I got this strange error (I added an ESP_LOGI line that prints the MAC just to know if I was getting any of the devices I wanted):

[image]

ajpaezm, 2019-11-18:
@cyberman54, when you said:

> There some verbose debug options in macsniff.cpp. Use highest debug level setting in platformio.ini and check in console messages if the MAC is seen and listed as counted or dropped.

You mean there's an actual printed message saying "Counted" or "Dropped"? I set Verbose = 1 on paxcounter.conf and set the debug_level = 5 in the .ini file. I didn't see something like that, is there an additional line I should write to see what you're telling on the above message?

Thanks.

cyberman54, 2019-11-19:
Sniffed MACs ar either shown as "known" or "new".
If you don't see this, your radio either did not receive a MAC, or you have filter settings enabled.
You can display filtered MACs by uncommenting 
https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/macsniff.cpp#L128

cyberman54, 2020-11-30:
Continued in https://github.com/cyberman54/ESP32-Paxcounter/issues/675

closed 2020-11-30 by cyberman54