cyberman54/ESP32-Paxcounter #342: Proximity Payload not sending?

Zufallsfuchs, 2019-04-05:
We've enabled the beacon alarm function via remote command  0x11 0x01
And set the MAC to the address of a smartphone we used for testing via rcommand 0x12 01 ...

But sill we don't receive any responses concerning proximity in TTN nor in the serial monitor.

We are using the packed_decoder.js and the packed_converter.js available in the repo.

Are we missing a step or is there some problem with the code?

simzik, 2019-05-06:
Same Problem on my side. I also use an ESP8266 to constantly send Beacons every Second. But i do not get a beacon alarm.

cyberman54, 2019-05-06:
Did you use Wifi oder Bluetooth clients/beacons?

simzik, 2019-05-06:
I use wifi from the esp8266

cyberman54, 2019-05-06:
Sure this is possible. The OUI filter currently is applied before beacon match is processed (see macsniff.cpp). To ensure you MAC is not OUI filtered, compile the code without Vendorfilter.

cyberman54, 2019-10-22:
Fixed.

https://github.com/cyberman54/ESP32-Paxcounter/pull/475

closed 2019-10-22 by cyberman54