cyberman54/ESP32-Paxcounter #345: Issue getting TTGO V3 1.5 running

lephisto, 2019-04-06:
Hello,

trying to get 1.7.31 running on a TTGO V3 1.5 20180523:

```
[I][main.cpp:122] setup(): Starting PAXCNT v1.7.31
[I][main.cpp:135] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
[I][main.cpp:137] setup(): Internal Total heap 251148, internal Free Heap 225212
[I][main.cpp:143] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.3-beta1-179-ge931fe9f5-dirty
[I][main.cpp:145] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:146] setup(): Wifi/BT software coexist version 1.1.3
[I][main.cpp:150] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:156] setup(): Arduino LMIC version 2.3.2.0
[I][configmanager.cpp:186] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:52] open_storage(): Opening NVS
[I][configmanager.cpp:57] open_storage(): Done
[I][configmanager.cpp:200] loadConfig(): NVRAM settings version = 1.7.31
[I][configmanager.cpp:219] loadConfig(): bsecstate = 0
[I][configmanager.cpp:224] loadConfig(): lorasf = 9
[I][configmanager.cpp:232] loadConfig(): txpower = 15
[I][configmanager.cpp:240] loadConfig(): adrmode = 1
[I][configmanager.cpp:248] loadConfig(): screensaver = 0
[I][configmanager.cpp:256] loadConfig(): screenon = 1
[I][configmanager.cpp:264] loadConfig(): countermode = 0
[I][configmanager.cpp:272] loadConfig(): sendcycle = 30
[I][configmanager.cpp:280] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:288] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:296] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:304] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:312] loadConfig(): blescantime = 8
[I][configmanager.cpp:320] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:328] loadConfig(): rssilimit = 0
[I][configmanager.cpp:336] loadConfig(): payloadmask = -1
[I][configmanager.cpp:344] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:352] loadConfig(): Run mode = 0
[I][configmanager.cpp:359] loadConfig(): Done
[I][lorawan.cpp:372] lora_stack_init(): LORA send queue created, size 510 Bytes
[I][lorawan.cpp:374] lora_stack_init(): Starting LMIC...
FAILURE 
.piolibdeps/MCCI LoRaWAN LMIC library_ID5774/src/lmic/oslmic.c:53
```

I tried with ttgov21new and ttgov21old, nothing :(

cyberman54, 2019-04-06:
Please upgrade paxcounter software to current v1.7.5 and build code using `ttgov21old.h` then report result here again.

lephisto, 2019-04-06:
The result is still the same. :(

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
[I][main.cpp:131] setup(): Starting PAXCNT v1.7.5
[I][main.cpp:144] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
[I][main.cpp:146] setup(): Internal Total heap 271428, internal Free Heap 245708
[I][main.cpp:152] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.3-beta1-179-ge931fe9f5-dirty
[I][main.cpp:154] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:155] setup(): Wifi/BT software coexist version 1.1.3
[I][main.cpp:159] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:164] setup(): Arduino LMIC version 2.3.2.0
[I][macsniff.cpp:36] printKey(): DevEUI: XXX
[I][macsniff.cpp:36] printKey(): AppEUI: XXX
[I][macsniff.cpp:36] printKey(): AppKey: XXX
[I][configmanager.cpp:185] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:50] open_storage(): Opening NVS
[I][configmanager.cpp:55] open_storage(): Done
[I][configmanager.cpp:199] loadConfig(): NVRAM settings version = 1.7.5
[I][configmanager.cpp:217] loadConfig(): bsecstate = 0
[I][configmanager.cpp:222] loadConfig(): lorasf = 9
[I][configmanager.cpp:230] loadConfig(): txpower = 15
[I][configmanager.cpp:238] loadConfig(): adrmode = 1
[I][configmanager.cpp:246] loadConfig(): screensaver = 0
[I][configmanager.cpp:254] loadConfig(): screenon = 1
[I][configmanager.cpp:262] loadConfig(): countermode = 0
[I][configmanager.cpp:270] loadConfig(): sendcycle = 30
[I][configmanager.cpp:278] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:286] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:294] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:302] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:310] loadConfig(): blescantime = 8
[I][configmanager.cpp:318] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:326] loadConfig(): rssilimit = 0
[I][configmanager.cpp:334] loadConfig(): payloadmask = 255
[I][configmanager.cpp:342] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:350] loadConfig(): Run mode = 0
[I][configmanager.cpp:357] loadConfig(): Done
[I][lorawan.cpp:419] lora_stack_init(): LORA send queue created, size 510 Bytes
[I][lorawan.cpp:421] lora_stack_init(): Starting LMIC...
FAILURE 
.piolibdeps/MCCI LoRaWAN LMIC library/src/lmic/oslmic.c:53
```

cyberman54, 2019-04-07:
This error means that LMIC stack cannot initialize the LoRa radio chip.

Possible root causes:
- wrong pin mapping regarding SPI interface of LoRa chip -> double checl pin mappings
- wrong chip type -> try swap SX1272/1276 in board's hal file
- broken hardware

I tested ttgov2old.h on a TTGO T3 board, it works, but the pcb has not a label with printed date. Maybe you have a different version. It seems there are clones on the aliexpress market...

lephisto, 2019-04-07:
I beeped through the board and found the following wiring (assuming it's a Lora-C1 board used on that TTGO T3):

```
MISO -> ESP IO19
MOSI -> ESP IO27
SCK ->  ESP IO5
NSS ->  ESP IO18
NRESET -> ESP IO12
DIO0 -> ESP IO26
```
Interessting is that DIO1/2 are only wired to the Board headers, and not to the ESP. Do i need to bridge them to a free Port? 
```
DIO1 -> Lora1 (board)
DIO2 -> Lora2 (board)
```

lephisto, 2019-04-07:
Okay, addition, IO1/2 are connected to 32/33 with a little R in between..

cyberman54, 2019-04-07:
what is Lora-C1? Link / pic?

cyberman54, 2019-04-07:
You need DIO1 for LMIC and must wire it to a ESP32 GPIO.

lephisto, 2019-04-07:
https://www.banggood.com/LoRa1276-C1-SX1276-868MHz-LoRa-Module-Remote-Spread-Spectrum-Wireless-Module-20dBm-100mW-3-5KM-p-1416393.html?rmmds=search&cur_warehouse=CN

cyberman54, 2019-04-07:
All TTGO boards i have seen so far have HPD13 modules, should be compatible with Semtech SX1276

cyberman54, 2019-04-07:
try to change Reset pin to (12) in ttgo21old.h, all other settings seem to match.

lephisto, 2019-04-07:
I made an, eehm, graphic. Maybe it's normal, but like mentioned there's two Resistors between Chipout and ESP IN for DIO1 and DIO2.

[image]

lephisto, 2019-04-07:
Oh. Setting RST seemed to help. JOIN WAIT.

cyberman54, 2019-04-07:
If you see Join Wait, DIO1 is probably missing. Check again wiring.

cyberman54, 2019-04-07:
Those resistors are probably 0 Ohm bridges, used for SMD machines

lephisto, 2019-04-07:
I measured them bit 22.5 Ohms, each. Could Join Wait also mean, that the next Gateway is too far away? I don't have one local..

cyberman54, 2019-04-07:
Of course you need a LoRaWAN gateway in range, for the network which your keys are valid for. Otherwise you can't join.

cyberman54, 2019-04-07:
[Zeisstower](https://ttnmapper.org/colour-radar/?gateway=B827EBFFFE738665&type=alpha&hideothers=on) should do the job!?

lephisto, 2019-04-07:
Yes, exactly. It should be in range, roughly 2.5km. I configured all the keys and euis i needed. Will get in the car later on and check if I get data through..

cyberman54, 2019-04-07:
If it still does not work, try to change DIO1/DIO2 in the hal file.
(DIO2 is actually not needed)

#define LORA_IO1  (33) -> (32)
#define LORA_IO2  (32) -> (33)

lephisto, 2019-04-07:
Btw, how can you know it's Zeisstower? :)

cyberman54, 2019-04-10:
Does ist work now? Can we close this?

closed 2019-04-10 by cyberman54