cyberman54/ESP32-Paxcounter #353: Send GPS data with every measurement?

proffalken, 2019-04-16:
Hi,

I've noticed that because the GPS data is only sent on channel 4, correlating events (PAX/BME680 measurements etc) against GPS position is nigh-on impossible in InfluxDB backends unless you somehow store the value from channel 4 somewhere when it arrives, and then add it in via a custom script each time a packet from that device arrives with other sensor data in it.

Is it possible to configure the device so that it sends the GPS data *as well as* the sensor data?

For example, it would be great if I could get a payload that includes both the lat/long/altitude *and* the VOC level from the BME680 in the same packet so I can graph the VOC level against the GPS co-ordinates

cyberman54, 2019-04-16:
Spreading sensor values on different ports was done intentionally, to keep the payload as small as possible for different use cases.

If you need a combined payload it's very simple to modify the code. Just add methods of class payload (found in `payload.h`) as you need in `senddata.cpp`.

For example, to combine pax, VOC and GPS:

```
case COUNT_DATA:
      payload.reset();
      payload.addCount(macs_wifi, MAC_SNIFF_WIFI);
      payload.addBME(bme_status);
      payload.addGPS(gps_status);
      SendPayload(COUNTERPORT, prio_normal);
```

Be aware that you may see large delays when transmitting huge payloads on a LoRaWAN network, depending on duty cycle and spreading value of transmission.

closed 2019-04-16 by cyberman54

proffalken, 2019-04-16:
Magic, thank you!

I don't necessarily need all the metrics in the same payload, but having GPS to accompany the various other metrics is awesome.

cyberman54, 2019-04-16:
Remember: if you combine payload you will get delays when transmitting via LoRaWAN, so your GPS locations won't match the timestamps. This is a problem e.g. when used paxcounter device with ttnmapper (but, in general, this works)