cyberman54/ESP32-Paxcounter #357: On Lopy always rebooting after some seconds

kami83, 2019-04-23:
Hi,

i am running the Paxcounter on a LOPY, but with normal settings i can see on the Terminal, that it is always rebooting?

I (651) wifi: ic_enable_sniffer
[I][main.cpp:368] setup(): Features: ANT GPS LORA OUIFLT PACKED WIFI
[I][main.cpp:371] setup(): Starting Interrupt Handler...
[I][main.cpp:401] setup(): Starting Timers...
[I][main.cpp:450] setup(): Starting Timekeeper...
[I][timekeeper.cpp:109] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][lorawan.cpp:339] onEvent(): JOINING
[I][lorawan.cpp:339] onEvent(): JOIN WAIT
[D][cyclic.cpp:36] doHousekeeping(): IRQhandler 3416 bytes left | Taskstate = 0
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c:1504 (eTaskGetState)- assert failed!
abort() was called at PC 0x4008fc35 on core 1

Backtrace: 0x40092688:0x3ffb4aa0 0x400928b9:0x3ffb4ac0 0x4008fc35:0x3ffb4ae0 0x400d2eae:0x3ffb4b00 0x400d311b:0x3ffb4b20 0x4008df19:0x3ffb4b50

Rebooting...
ets Jun  8 2016 00:22:57

Can someone help me?

Thanks a lot

BG kami83

cyberman54, 2019-04-23:
Did you disable the GPS settings in lopy.h? They can only be used if your lopy is plugged on a expansion board. If not disable them (place // in front of each line)

kami83, 2019-04-23:
Hi,

thanks a lot. This works fine. 

Cu kami83

closed 2019-04-23 by kami83