cyberman54/ESP32-Paxcounter #359: The oled screen does not turn on.

reisabdullahi, 2019-05-03:
Hi,

I managed to run this thank to your explanation. I have a Heltechv2 board and when I build I get 2 errors I think this is why the screen does not turn on.

".piolibdeps/MCCI LoRaWAN LMIC library_ID5774/src/hal/getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
.piolibdeps/MCCI LoRaWAN LMIC library_ID5774/src/hal/getpinmap_thisboard.cpp:51:72: note: #pragma message: Board not supported -- use an explicit pinmap
#pragma message("Board not supported -- use an explicit pinmap")
^"

I tried to build the project with ttgov2 and I was getting the same error. 

PS: Tried some other sample code the screen works.

cyberman54, 2019-05-03:
The above compiler messages are normal and will not cause your display problem.

For solution look here and double check that you really own a Heltec V2 board, not a V1.
https://github.com/cyberman54/ESP32-Paxcounter/issues/311

reisabdullahi, 2019-05-08:
Thank you, it seems connecting the pins worked for me also. Quite strange, when I run a sample code the screen turns on without having to connect the pin to the ground. 

Thank you :)

reisabdullahi, 2019-05-16:
Hi, 

I was stuck for a bit, my Heltec is V2 (at least that's what it is written on it), after posting here only  with the V1 code the was screen turning on (and connecting one pin to the ground) I was getting PAX results but I could not ping the LoRa Gateway. So in the end I edited the oled PIN configuration in hltecv2.h used the oled pin layout defined in heltec.h. And now I finally managed to get the screen working and just pinged the gateway. I will continue looking into this.

Can it be that heltec just changed the pinlayout in V2 again?

cyberman54, 2019-05-16:
There was a bug in platformio.ini causing that for heltec v2 the pin mapping of v1 was used.
I fixed this today. Please check if this solves your problem.

https://github.com/cyberman54/ESP32-Paxcounter/commit/6796d65

cyberman54, 2019-05-19:
Should be solved now (see above).
If problems persist, please re-open.

closed 2019-05-19 by cyberman54