cyberman54/ESP32-Paxcounter #364: usage of fport for payload in timesync protocol

FlorianLudwig, 2019-05-11:
I like to suggest moving the sequence number from the fport field into the payload for three reasons:

 * **Readability of code and protocol**  The port field is 
 * **Forward compatibility** It might complicate the implementation later on.
 * **Security**  The fport field is not encrypted while the payload is. This opens the APPSKey to known-plain-text attacks. While there are no (known) practical attacks of this kind against AES (yet) I would still consider this a design fault.  This could also be avoided by moving the seq no in the upstream message `time query` into the fport but that would be especially bad for forward compatibility since it complicated server implementation to handle different versions quite a bit.

I would like to argue that the benefit of saving one byte downstream does not outweigh the points above.  Shrinking the payload of a LoRaWAN package from 6 to 5 bytes:
 * SF7 (51ms) no change in airtime 
 * SF9 (185ms) no change in airtime 
 * SF12: decrease from 1319ms to 1155ms (-164ms, 12%)

cyberman54, 2019-05-11:
Ok, agreed. Only goal to use FPort for Seq no. was to save 1 byte of payload.

cyberman54, 2019-10-04:
Will be fixed in next version.

closed 2019-10-04 by cyberman54