cyberman54/ESP32-Paxcounter #365: Proposal: add timezone and daylight saving handling

FlorianLudwig, 2019-05-13:
Add an optional field for timezone offset to UTC to the time-sync lorawan protocoll to allow for timezone and daylight saving handling.

Encoding: 8 bit singed int, encoding offset to utc in multiples of 15 minutes [0]

As neither timezone nor daylight saving time changes often the field in the downlink message should be optional.  Some heuristic on the backend side is needed when to send this extra bit of information:

- include it on first time sync response after a reboot (make reboot detestable server side)
- if requested (flag on upstream message)
- include once per day
- include on change (dst)
- It might be worthwhile to have an additional field for "will change", to announce DST changes ahead of time to nodes. Probably takes another 5 bytes (4 bytes UTC time when it will happen and 1 byte new offset)

[0] Timezones are weird so I would not bet there was or will ever be a timezone that is not a multiple of 15 minutes. But so far I have not found any instance of a timezone that was not encodable with 15 minutes multiples. Also currently this (should work)[https://en.wikipedia.org/wiki/List_of_UTC_time_offsets] and i don't see any other generic way of encoding timezone without wasting more bandwidth.

FlorianLudwig, 2019-08-09:
@cyberman54 Any thoughts on this?

cyberman54, 2019-08-11:
Could be a ninfty enhancement. Makes senses only if we add a full featured timeserver to the repository, i think.

cyberman54, 2019-10-04:
Will be fixed in next version.

closed 2019-10-04 by cyberman54