cyberman54/ESP32-Paxcounter #370: Timesync protocol: LoRa transfer time is not considered

cyberman54, 2019-05-25:
`process_timesync_req()` in `timesync.cpp` does not consider air transmit time of the timesync packet. Not sure if this should be considered. If yes, we need to check LoRa SF for the timesync packet and convert SF to milliseconds air time delay.

FlorianLudwig, 2019-05-25:
As long as the time on the gateway and on the node is taken on the same time (both at the start or at the end of the frame) the airtime does is not part of the calculation.

In the implementations I know of the gateway timestamps packages at the end of the transmission, so does the current node implementation.  Therefor the airtime is not part of the calculation.

cyberman54, 2019-05-25:
Ok, so this seems not to be a source of erraneous timesyncs.
Thanks.

closed 2019-05-25 by cyberman54