cyberman54/ESP32-Paxcounter #373: Problem when compiling TTG T-BEAM

Tomtendo, 2019-05-26:
Hi Folks,

when trying to compile the Paxcounter i always get this Message: 

.piolibdeps/MCCI LoRaWAN LMIC library/src/lmic/lmic_config_preconditions.h:92:52: fatal error: ../../../../../src/lmic_config.h: No such file or directory

Thanks for the Help!

cyberman54, 2019-05-26:
Read the hints in platformio.ini regarding the platformio version and select appropriate path, or update your platformio version.

Unfortunately this is a platformio problem which came up with last platformio update.

shenzou, 2019-05-28:
Hi @Tomtendo ,
I had the same problem, check that your platformio version is not under 4.0.0a15. If it is, uncomment the following line:
'-DARDUINO_LMIC_PROJECT_CONFIG_H=../../../src/lmic_config.h'
and comment the following line:
'-DARDUINO_LMIC_PROJECT_CONFIG_H=../../../../../src/lmic_config.h'
in the platformio.ini file, as explained on it.

cyberman54, 2019-05-29:
@shenzou @Tomtendo I found a way to avoid this, regardless of platformio version. Is currently coded in branch development. Could you, please, re-test with the development branch, and report results here? Thanks.

Please note your environment: operating system (Windows / Linux) and editor (Atom / VSC). Thanks.

Tomtendo, 2019-05-29:
It worked for me by using an old Platformio Version, thank you guys for the quick help.

cyberman54, 2019-05-29:
Solved.

closed 2019-05-29 by cyberman54