cyberman54/ESP32-Paxcounter #374: Question concerning WiFi / Voltage Messages

shinebar1001, 2019-05-27:
Thank you for your great work. I've got a question concerning the sending of the messages. When the vendor filter is activated, paxcounter is sending "voltage" message on port 8 quite often and messages with counted WiFi-Devices don't get transmitted. Is there a way to supress the voltage-messages? Am I doing something wrong?
best regards
Danny

[image]

shinebar1001, 2019-05-27:
Question answered: I removed the definition of the battery capabilities from the HAL. But as I use the TTGO 2.1 V 1.6 USB-powered there must be something wrong with the internal divider - any suggestions on making this slightly more configurable?

closed 2019-05-27 by shinebar1001

cyberman54, 2019-05-27:
@shinebar1001 If you plug USB you cannot measure battery voltage on the TTGO.

shinebar1001, 2019-05-30:
Yep, of course - but out of the box paxcounter tried to do that. I noticed a picture where "USB" was displayed - am I doing something wrong? Otherwise I'd try to submit a PR which makes this configurable

cyberman54, 2019-05-30:
Please make PR, thanks.

shinebar1001, 2019-05-30:
Got an idea already, will take some time as I only got one device which is currently in the office.

cyberman54, 2019-05-31:
@shinebar1001 
Maybe you can solve issue #321 with your PR, too?

Please have in mind that on some boards there is a Vext on/off switch, e.g. EcoPower, HeltecV2 and TTGP v1.6

I found out that setting of Vext on/off switch may or must be considered when it comes to measure voltage of battery. Depending on schematic of boards.

shinebar1001, 2019-05-31:
That seems possible, I just ordered a HeltecV2