cyberman54/ESP32-Paxcounter #376: Time source only changes after reset.

mpxgit, 2019-05-30:
On a T-Beam v07 the timesource indicator stays on "?" after the GPS already has a fix.
Pressing the reset button the time is synced from the GPS and the indicator switches to "G".
During my tests there is no LoRaWan gateway in range.

cyberman54, 2019-05-30:
If the device has no valid time, it tries to sync time every `TIME_SYNC_INTERVAL_RETRY` (set in paxcounter.conf) minutes.

Default ist 10 minutes. Did you wait 10 minutes?

mpxgit, 2019-05-30:
Yes, even after several hours it stays on the "?". Even when switching the Pages the GPS page is OK and has a correct LAT / LON. Back to the Main Page time is still out of sync.

cyberman54, 2019-07-06:
I still cannot reproduce this on TTGO T-Beam v7.
Can you give a more precise description? Does this only occur after startup?

mpxgit, 2019-07-06:
1. Power on the TTGO
2. wait for sat fix 
3. numbers of stats displayed but timesource stays at „?“
4. Press reset button after boot timesoure is correct immediately. 

So this only occurs after a „cold Start“

cyberman54, 2019-07-06:
I could reproduce this now on a TTGO T-Beam v7, you are right.
So far i have no idea what causes this bug, i assume it's a behaviour of the used TinyGPS+ library.

cyberman54, 2019-07-07:
Fixed.
Thanks for reporting this bug.

closed 2019-07-07 by cyberman54