cyberman54/ESP32-Paxcounter #377: Vane Display / Fallblatt Anzeigen

sbamueller, 2019-05-30:
Paxcounter needs to support “Fallblattanzeigen“ a popular way of announcing train departures, flights or even stock quotes. 
Maybe so called flipdots are a solution: https://flipdots.com/en/home/

Tutorial: https://create.arduino.cc/projecthub/iizukak/flip-dot-clock-3dd850

cyberman54, 2019-05-30:
You're welcome to deliver a pull request for this awesome feature!

chopmann, 2020-04-02:
Most of the "old ones" in Germany speak IBIS Wagenbus Protocol and run on a 24V Bus System. If the controller is still working, is just a matter of sending a string.

chopmann, 2020-04-02:
https://www.vdv.de/telegrammvergleichstabelle-ibis-zu-ibis-ip-v1.0.pdfx
Telegram Nr 3 and and 003a

closed 2020-11-21 by cyberman54