cyberman54/ESP32-Paxcounter #380: Error when trying to use heltec / heltec v2

sbamueller, 2019-06-08:
`Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing heltec (framework: arduino; platform: espressif32@1.8.0; board: heltec_wifi_lora_32)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
AttributeError: 'module' object has no attribute 'load_project_config':
  File "/Users/sbamueller2/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py", line 113:
    env.SConscript(item, exports="env")
  File "/Users/sbamueller2/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/sbamueller2/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/Users/sbamueller2/Downloads/ESP32-Paxcounter-master 2/build.py", line 14:
    project_config = util.load_project_config()`

sbamueller, 2019-06-08:
Problem occours as well under Atom and with TTGO Board

cyberman54, 2019-06-23:
actually these are two issues with different root ause.
Both now solved; bumped to new version 1.7.62
Will require PIO v4 from now on.

closed 2019-06-23 by cyberman54