cyberman54/ESP32-Paxcounter #384: TTGO beam cannot find LoRa module

TD-er, 2019-06-25:
I have been trying to get my TTGO beam module to work, but until now I am not successful.

N.B. I do have the T22_v07 board.

[image]

This board does mention in the silkscreen that there should be some connection between Lora1 and GPIO33 and Lora2 and GPIO32.
But just to be sure, I also tested with wires between those pins.

It simply does not get past this point (the assert):
```
[I][lorawan.cpp:424] lora_stack_init(): LORA send queue created, size 510 Bytes
[I][lorawan.cpp:426] lora_stack_init(): Starting LMIC...
FAILURE
.piolibdeps\MCCI LoRaWAN LMIC library\src\lmic\oslmic.c:53
```

The related function triggering the assert on `oslmic.c:53` :
```c++
void os_init() {
    if (os_init_ex((const void *)&lmic_pins))
        return;
    ASSERT(0);
}
```

Things that I find odd are that I don't see any definition of the pins for this board.
See https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/hal/ttgobeam.h
This one does not have defines like the ones that are present in `src/hal/ttgov2.h` and some others.
I mean pin configs like these:
https://github.com/cyberman54/ESP32-Paxcounter/blob/833cd4204e284031c0794a8c310f0aa0ef4ae07f/src/hal/ttgov21new.h#L31-L39

I also added these defines to my `ttgobeam.h` just to be sure it is defined.

Does anyone have other suggestions on how to get this board to work?

cyberman54, 2019-06-25:
Pin definition for TTGO T-Beam comes from pins_arduino.h which is part of the ardunio-espressif32 framework repo. check the file in this repo and compare settings with your hardware. 

The error message typically points to a non working SPI interface, that means the LoRa chip is not seen by the Lmic stack.

TD-er, 2019-06-25:
> The error message typically points to a non working SPI interface, that means the LoRa chip is not seen by the Lmic stack.

That's also what I was thinking, but how can we make it 'visible' ?
Like I said, I added wires (which do not seem necessary judging by the silkscreen), added the defines to make sure and also applied some extra solder to the pads to make sure they make contact with the PCB.

cyberman54, 2019-06-25:
This is the source from where the pin mapping are fetched:

https://github.com/espressif/arduino-esp32/blob/master/variants/t-beam/pins_arduino.h

cyberman54, 2019-06-25:
I retested the current code in master dev with a TTGO T-Beam board which is labelled as yours.
Works here.

Maybe it's an issue with your hardware, or they changed something.

TD-er, 2019-06-25:
OK, thanks.

I think this board is maybe defective then.
I will take it with me to the TTN meetup tomorrow and maybe Jac or Leonel can see what's wrong here.

cyberman54, 2019-06-27:
Closed, since retest showed code is running on a TTGO T-Beam v7.
If you find something code related for this issue, please re-open.

closed 2019-06-27 by cyberman54

TD-er, 2019-06-27:
Yesterday at the TTN Meetup we tested with 3 of these boards.
One T22-v07 like mine, a T22-v05 (somewhat inbetween version actually, a pre-production build) and my board.
The other two worked just fine with the test code we used. So the board appears to be a DOA.
Too bad it took so many hours to try and make it work. :(

cyberman54, 2019-06-27:
I read some DOA reports on the net for TTGO boards.

I also saw reports that state TTGOs to have a wrong labelled LoRa chip, which is not SX1276, but SX1272. You could try this by changing in the hal file

`#define CFG_sx1276_radio 1 // HPD13A LoRa `

to 

`#define CFG_sx1272_radio 1`

TD-er, 2019-06-28:
@cyberman54 
Thanks for the suggestion. Will try that tomorrow.

TD-er, 2019-06-29:
@cyberman54 
Nope, too bad, it wasn't the fix I hoped for.
```
FAILURE
.piolibdeps\MCCI LoRaWAN LMIC library\src\lmic\oslmic.c:53
```

I think we should consider this one as a very big ESP32 board with GPS and 18650 power supply.