cyberman54/ESP32-Paxcounter #388: Error: Please specify platform for 'ota' environment

Greust, 2019-07-01:
Hi,
first of all: thanks for your work!

I successfully use the Paxcounter on a TTGO 2.1 Board. 
Today i tried to use the new Version on a Heltec V2, but Visual Studio aswell as Atom end the Cleanup with the error:

ERROR: Please specify platform for 'ota' environment

Any idea how to fix this?

Thanks a lot!

cyberman54, 2019-07-01:
Please update to current version of respository. This should be fixed already.

tobievii, 2019-07-09:
its not , just compiled does not work

cyberman54, 2019-07-09:
@tobievii i can't reproduce this here, please state to be sure we have same base:
- which platformio version do you use (pio --version) 
- which #commit of master branch did you compile

cyberman54, 2019-07-09:
Reopened.

cyberman54, 2019-07-15:
Probably this issue is platformio v3.x related; meanwhile i got confirmations that the code compiles fine with platformio v4.x.

So please make sure you're using pio v4.x,  not 3.x.

If problem persist with pio v4.x, please reopen this issue and try to give some more context information. thanks.

closed 2019-07-15 by cyberman54