cyberman54/ESP32-Paxcounter #390: How to set transmission interval

danielknox, 2019-07-10:
Hi!
I'm using the latest dev build of platform io. The program compiles and uploads fine to a lopy4; however, I'm finding the send interval over lorawan is around every 60seconds (wifi pax count + message concerning voltage state). I've changed SENDCYCLE in paxcounter.conf  to 150 (for approximately 5 minute transmission intervals?) and clean and built it in again in platformio. However the device still sends every minute. Is there another setting somewhere I'm missing for this? Similarly what is the relationship between send interval and the length of time a salted MAC address is retained in memory? 
Cheers!

cyberman54, 2019-07-10:
Please read 
https://github.com/cyberman54/ESP32-Paxcounter/issues/272
how to properly change device settings during runtime and changing defaults in the code.

Send interval (parameter SENDCYCLE in paxcounter.conf) also sets the memory duration, since memory is cleared after each send cycle.

closed 2019-07-10 by cyberman54

danielknox, 2019-07-10:
Cheers! That makes a lot of sense now!

ajpaezm, 2019-10-22:
Hi @cyberman54 and thanks @danielknox for opening this issue because I had the same doubt. 

SENDCYCLE has a limit then (around 4.20 minutes wait time), right? Or is it 8.40 minutes total? There's no other way to enhance this capability?

If the answer is "no", then, how difficult could be to use millis() and a predefined PERIOD to:

1) Send a counter of BLE/WIFI scanned (2 byte array) once every minute.
2) Send HASH once every 5 minutes.

cyberman54, 2019-10-23:
@ajpaezm You're welcome to suggest your PR to enhance this!

cyberman54, 2019-10-23:
@ajpaezm i looked into the code. Easiest option would be to interprete SENDCYCLE as minutes or minutes/2, instead of seconds/2. 

This way we could have intervals from 1 ... 255 minutes or 2 ... 255 minutes.

Intervals less than 2 minutes probably make no sense because of duty cycles and network fair use policies.

The real issue here is, that send cycle is identical with scan cycle. This should be reconsidered, and will probably result in an more extensive code refactoring.