cyberman54/ESP32-Paxcounter #391: control pin output via platormio.ini

sbamueller, 2019-07-15:
it would be helpful if the pin Output was controled via platformio.ini.

As pin mapping for ttgo t-beam I'd like to suggest:
`
// Pin numbers for TTGO T-Beam
// see https://twitter.com/RecumbentTravel/status/1150471613617442819
// 36 labeled VP on T-Beam see: https://twitter.com/RecumbentTravel/status/1150472470132142090
// OE = EN pin on some displays 
// Connect GND on T-Beam to GND on display, connect 5V on T-Beam to 5V on display

#define MATRIX_DISPLAY_SCAN_US          500     // Matrix display scan rate in microseconds (1ms is about 'acceptable')
#define LED_MATRIX_LATCHPIN             13      // Connects to LAT pin on display
#define LED_MATRIX_CLOCKPIN             04      // Connects to CLK pin on display
#define LED_MATRIX_EN_74138             33      // Connects to OE or EN pin on display
#define LED_MATRIX_LA_74138             14      // Connects to LA pin on display
#define LED_MATRIX_LB_74138             32      // Connects to LB pin on display
#define LED_MATRIX_LC_74138             25      // Connects to LC pin on display
#define LED_MATRIX_LD_74138             36      // Connects to LD pin on display 
#define LED_MATRIX_DATA_R1              34      // Connects to R1 pin on display`

cyberman54, 2019-07-15:
Standard pin mappings for boards should always some from single point of truth `pins_arduino.h`, found [in arduino-espressif32 repository](https://github.com/espressif/arduino-esp32/tree/master/variants).

Non standard pin mappingst, e.g. for connecting external hardwire like BME680 or LED_MATRIX should be added to the board files in hal directory.

So the task is to move LED_MATRIX mappings from `paxcounter.conf` to board file(s).

cyberman54, 2019-07-15:
Done.

closed 2019-07-15 by cyberman54

cyberman54, 2019-07-15:
@sbamueller look at https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/hal/wemos32matrix.h how to do it