cyberman54/ESP32-Paxcounter #392: Wifiscan broke in last version

IoTAdri, 2019-07-15:
updated to last version today (for Platformio 4.0)
now my Heltec does not show the wifichannel he scans anymore just a star (no problem).. 
but no devices show up.. counts nothing.. 
(did try on 3 seperate boards Heltec V1, V2 and TTGo-beam).. reports 0 in my TTN-console

also Bluetooth does not count or report anything although i set BLECOUNTER to 1
(problem was there before this version on all three boards)

-last version had problems joining, is fixed now (sets SF9 after joining instead of SF7 )-

PlatformIOcore 4.0, ESP32 1.9, VSC 

Please advise..

cyberman54, 2019-07-15:
I reverted the commit which probably caused this problem, although i cannot reproduce it.
Can you, please, re-test? Thanks.

IoTAdri, 2019-07-15:
Wifi-channelcounter re-appeared but no counting Wifi nor BL
but i saw something..
in LoadConfig -> wifiantenna = 0 and BLEScanmode = 0
how do i reset the config? 

=============================

// Set this to include BLE counting and vendor filter functions, or to switch off WIFI counting
#define VENDORFILTER                    1       // set to 0 if you want to count things, not people
#define BLECOUNTER                      1       // set it to 1 if you want to use BLE count, at expense of power & memory
#define WIFICOUNTER                     1       // set it to 0 if you want to switch off WIFI count

========================

[I][main.cpp:137] setup(): Starting PAXCNT v1.7.8
[I][main.cpp:150] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
[I][main.cpp:152] setup(): Internal Total heap 249952, internal Free Heap 224444
[I][main.cpp:158] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2-18-g977854975
[I][main.cpp:160] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:161] setup(): Wifi/BT software coexist version 1.1.5  
[I][main.cpp:165] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:170] setup(): Arduino LMIC version 2.3.2.51  
[I][macsniff.cpp:36] printKey(): DevEUI: xxxremovedxxx
[I][macsniff.cpp:36] printKey(): AppEUI: xxxremovedxxx
[I][macsniff.cpp:36] printKey(): AppKey: xxxremovedxxx
[I][configmanager.cpp:185] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:50] open_storage(): Opening NVS
[I][configmanager.cpp:55] open_storage(): Done
[I][configmanager.cpp:199] loadConfig(): NVRAM settings version = 1.7.8
[I][configmanager.cpp:217] loadConfig(): bsecstate = 0
[I][configmanager.cpp:222] loadConfig(): lorasf = 9
[I][configmanager.cpp:230] loadConfig(): txpower = 15
[I][configmanager.cpp:238] loadConfig(): adrmode = 1
[I][configmanager.cpp:246] loadConfig(): screensaver = 0
[I][configmanager.cpp:254] loadConfig(): screenon = 1
[I][configmanager.cpp:262] loadConfig(): countermode = 0
[I][configmanager.cpp:270] loadConfig(): sendcycle = 60
[I][configmanager.cpp:278] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:286] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:294] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:302] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:310] loadConfig(): blescantime = 8
[I][configmanager.cpp:318] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:326] loadConfig(): rssilimit = 0
[I][configmanager.cpp:334] loadConfig(): payloadmask = 255
[I][configmanager.cpp:342] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:350] loadConfig(): Run mode = 0
[I][configmanager.cpp:357] loadConfig(): Done
[I][main.cpp:218] setup(): Starting LED Controller...
[I][lorawan.cpp:424] lora_stack_init(): LORA send queue created, size 530 Bytes
[I][lorawan.cpp:426] lora_stack_init(): Starting LMIC...
[I][main.cpp:348] setup(): Starting Wifi...
I (4938) wifi: wifi driver task: 3ffbb854, prio:23, stack:3584, core=0

cyberman54, 2019-07-15:
wifiant=0 is only important for devices with two antennas, like LoPy/LoPy.

Reset NVS config can be done by increasing version number in platformio.ini and reflash, or send remote command 09 02 to device

cyberman54, 2019-07-15:
Did you select the proper board in `platformio.ini` ?

```
[board]
;halfile = generic.h
;halfile = heltec.h
halfile = heltecv2.h
;halfile = ttgov1.h
;halfile = ttgov2.h
```

IoTAdri, 2019-07-15:
yepp... changed to the proper board everytime...

it seems increasing the versionnumber does not erase the config... setting the wifiant =1 and blescanmode =1 in the default seems to work...

i changed the versionnumber but in de loginfo no call to eraseconfig is to be seen...

on which port should i send 09 02 ? (this should also trigger the log so i should see it)

IoTAdri, 2019-07-15:
sent the 09 02 on port 02 and then the logfunction shows "eraseconfig" (but not on uptick from versionnumber)

cyberman54, 2019-07-15:
If you change versionnumber in platformio.ini, you need to rebuild and reflash code. After device starts up with new version, it will detect version migration and then reset to factory settings once.
If you send rcommand 09 02 on port 02 this triggers the same process.

But i doubt this issue is caused by nvs settings.

I re-tested the current code on a TTGO T-Beam v7, don't see any issues, seems to work, joins and counts.

IoTAdri, 2019-07-15:
I have updated the version from 1.7.8 several times until 1.7.11 but not once i saw "version set to " come by.... i think it is in the configfile... only when i set wifiant an blescanmode in the defaultvalues to 1 it works [workaround]

cyberman54, 2019-07-15:
very weird...

IoTAdri, 2019-07-15:
I think there are some ommisions in the  procedure DefaultConfig in configmanager.cpp.
several lines are wrong (should alline with paxcounter.conf): 
cfg.blescan = 1;  should be cfg.blescan = BLECOUNTER;
cfg.vendorfilter = 1; should be cfg.vendorfilter = VENDORFILTER;
etc.

a lot of varaibles from paxcounter.conf are not implemented

cyberman54, 2019-07-15:
Can you make pull request with solution?

IoTAdri, 2019-07-15:
will try (I am noob with Git)... 
(but this is a correction and works for the workaround, still not a solution for the "migrateVersion"-problem)

IoTAdri, 2019-07-15:
Procedure defaultConfig
configmanager.cpp:36 you put PROGVERSION (which is the NEW version) in cfg.version; 

Procedure loadConfig
configmanager.cpp:185 you first call defaultConfig (loading PROGVERSION in cfg.version)
configmanager.cpp:201 you compare cfg.version with PROGVERSION and if it is different (impossible) migrateVersion or even eraseConfig is called (this will never happen...)

will try to correct in the comming days...

cyberman54, 2019-07-15:
configmanager.cpp:198 populates cfg.version with value from flash, so we can detect a firmware update

cyberman54, 2019-07-16:
@IoTAdri Please note, below settings in paxcounter.conf are compiler switches (not runtime settings) with purpose to keep the flashbinary small for boards with limited flash.

```
#define VENDORFILTER                    1       // set to 0 if you want to count things, not people
#define BLECOUNTER                      0       // set it to 1 if you want to use BLE count, at expense of power & memory
#define WIFICOUNTER                     1       // set it to 0 if you want to switch off WIFI count
```

cyberman54, 2019-07-22:
Meanwhile i could reproduce this problem on an TTGO V2.1 (old version) and on a EBOX Widora module, but not on TTGO T-Beam.

The device stops display refresh and send cycle after initial syncing time and reporting a i2c semaphore access problem. This looks like the application interrupt stopped working for some strange reason:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9232
load:0x40080400,len:6412
entry 0x400806a8
[I][main.cpp:137] setup(): Starting PAXCNT v1.7.82
[I][main.cpp:150] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB embedded Flash
[I][main.cpp:152] setup(): Internal Total heap 265956, internal Free Heap 239860
[I][main.cpp:158] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2-18-g977854975
[I][main.cpp:160] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:161] setup(): Wifi/BT software coexist version 1.1.5
[I][main.cpp:165] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:170] setup(): Arduino LMIC version 2.3.2.51
[I][macsniff.cpp:36] printKey(): DevEUI: D8A01D4046F0FEFF
[I][macsniff.cpp:36] printKey(): AppEUI: 70B3D57ED00085F3
[I][macsniff.cpp:36] printKey(): AppKey: 309BF024F7562FCAE11D811A4F87C237
[I][configmanager.cpp:185] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:50] open_storage(): Opening NVS
[I][configmanager.cpp:55] open_storage(): Done
[I][configmanager.cpp:199] loadConfig(): NVRAM settings version = 1.7.82
[I][configmanager.cpp:217] loadConfig(): bsecstate = 0
[I][configmanager.cpp:222] loadConfig(): lorasf = 9
[I][configmanager.cpp:230] loadConfig(): txpower = 15
[I][configmanager.cpp:238] loadConfig(): adrmode = 1
[I][configmanager.cpp:246] loadConfig(): screensaver = 0
[I][configmanager.cpp:254] loadConfig(): screenon = 1
[I][configmanager.cpp:262] loadConfig(): countermode = 0
[I][configmanager.cpp:270] loadConfig(): sendcycle = 30
[I][configmanager.cpp:278] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:286] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:294] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:302] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:310] loadConfig(): blescantime = 8
[I][configmanager.cpp:318] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:326] loadConfig(): rssilimit = 0
[I][configmanager.cpp:334] loadConfig(): payloadmask = 255
[I][configmanager.cpp:342] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:350] loadConfig(): Run mode = 0
[I][configmanager.cpp:357] loadConfig(): Done
[I][lorawan.cpp:423] lora_stack_init(): LORA send queue created, size 530 Bytes
[I][lorawan.cpp:425] lora_stack_init(): Starting LMIC...
[I][main.cpp:348] setup(): Starting Wifi...
I (4935) wifi: wifi driver task: 3ffc383c, prio:23, stack:3584, core=0
I (10047) wifi: wifi firmware version: 6d404d4
I (10049) wifi: config NVS flash: disabled
I (10053) wifi: config nano formating: disabled
I (10058) wifi: Init dynamic tx buffer num: 32
I (10061) wifi: Init data frame dynamic rx buffer num: 10
I (10066) wifi: Init management frame dynamic rx buffer num: 10
I (10072) wifi: Init static rx buffer size: 1600
I (10077) wifi: Init static rx buffer num: 8
I (10080) wifi: Init dynamic rx buffer num: 10
I (10085) wifi: set country: cc=EU␀ schan=1 nchan=13 policy=1

W (10090) wifi: ap start fail

I (10151) wifi: ic_enable_sniffer
[I][main.cpp:362] setup(): Starting Interrupt Handler...
[I][main.cpp:392] setup(): Starting Timers...
[I][main.cpp:453] setup(): Starting Timekeeper...
[I][timekeeper.cpp:109] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][timesync.cpp:41] send_timesync_req(): [5.056] Timeserver sync request started
[I][main.cpp:460] setup(): Features: OTA LORA FILTER OLED PACKED WIFI
[I][lorawan.cpp:338] onEvent(): JOINING
[I][lorawan.cpp:223] onEvent(): DEVaddr=260120B0
[I][lorawan.cpp:338] onEvent(): JOINED
[I][lorawan.cpp:338] onEvent(): TX TIMESYNC
[I][lorawan.cpp:401] lora_send(): 1 byte(s) sent to LoRa
[I][lorawan.cpp:252] onEvent(): Received 5 bytes of payload, RSSI 11 SNR 9
[I][lorawan.cpp:338] onEvent(): TX COMPLETE
[W][timesync.cpp:117] process_timesync_req(): [16.611] Timesync handshake error: irq / i2c masking failed
[I][lorawan.cpp:338] onEvent(): TX START
[I][lorawan.cpp:401] lora_send(): 1 byte(s) sent to LoRa
[I][lorawan.cpp:338] onEvent(): TX COMPLETE

```

cyberman54, 2019-07-22:
@IoTAdri I think i fixed it. It was a masked interrupt which could happen in an edge case, no context with configmanager.cpp (which may be weird, but isn't buggy).
New code is in development branch.
Could you please rebuild, retest and report result here? Thanks.

cyberman54, 2019-07-24:
fixed, now also in master branch.
closed.

closed 2019-07-24 by cyberman54