cyberman54/ESP32-Paxcounter #403: GPS altitude is decoded as uint16 but should be int

TD-er, 2019-07-26:
See here, where the GPS altitude is decoded:
https://github.com/cyberman54/ESP32-Paxcounter/blob/3648bc794a724dcbdccbd3ce8e4c07da41e3c208/src/gpsread.cpp#L74

In the [decode functions](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/TTN/packed_decoder.js) it is being interpreted as uint16_t.
So, here in the Netherlands I am at 65533 altitude :)

cyberman54, 2019-07-27:
Thanks for reviewing and improving this!
Can you make PR for increase of resolution?

TD-er, 2019-07-27:
I can, but that also requires a change on the ESP side.
This will break compatibility. Is there a way to manage versions, or should we consider the old one non-working anyway and just go for it?

cyberman54, 2019-07-27:
I made a suggestion, see in development branch, could you please check it? (i can't since i am +65m over sea level).

https://github.com/cyberman54/ESP32-Paxcounter/commit/a3f036c4d2a2c6b86e0c0ba072b66090c5f8e5cc

And yes, it would be a hard breaking change. Don't know how to handle this.
There is already a second breaking change pending, but not yet deployed, see https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/lorawan.cpp#L72

TD-er, 2019-07-27:
Code looks good (had to check the meters were floating point, but that's fine :) )
I will now check it on a running node.

You can always declare the 'breaking changes' as v2.0, or introduce a new port number for it.
You can also add a single byte for generic system info in every message. This then changes the message length so the decoder can remain compatible.
In this byte you can encode some kind of version info of the protocol.
For example 3 bits for the version (can be different per port) and then you have 5 bits left for other stuff.

TD-er, 2019-07-27:
Seems to work just fine:
```json
{
  "altitude": -6.5,
  "hdop": 5.41,
  "latitude": 53.2xxxx,
  "longitude": 6.6xxxx,
  "sats": 4
}
```

I've also seen  -5.8

cyberman54, 2019-07-27:
Once again i made some changes, could you please test again
https://github.com/cyberman54/ESP32-Paxcounter/commit/3058973f667f6304231b440a20700a4aadda0759
if you get ninfty altitude values? Thanks!

TD-er, 2019-07-27:
Where did you make the changes?
The one you link only has an include added.

cyberman54, 2019-07-27:
Note: i moved calculation of altitude payload from `gpsread.cpp` to `payload.cpp`:

https://github.com/cyberman54/ESP32-Paxcounter/blob/development/src/payload.cpp#L95
https://github.com/cyberman54/ESP32-Paxcounter/blob/development/src/payload.cpp#L198

cyberman54, 2019-07-28:
Made another edit, hope this is it now.

https://github.com/cyberman54/ESP32-Paxcounter/commit/88db2cca2bf2a2d435af7a5164cf93554e7cf7fd

TD-er, 2019-07-28:
I am now looking at the payload.cpp file and I have a few remarks:
- Uptime is sent in the satus as a 64 bit value, but it is sent in seconds, not msec. (may take 68 year to overflow `int32_t` ) So you're using 4 extra bytes which are not needed.
- Sending uptime does call `intToBytes` function using 8 bytes, but using the >> operator to shift more bits than the size of the parameter is undefined behavior. See also [this StackOverflow question](https://stackoverflow.com/questions/16122512/what-happens-if-we-bitwise-shift-an-integer-more-than-its-size)
- An unsigned 32 integer will not be sent correctly (function `intToBytes` casts to `int32_t`)
- Not sure if any int value less than 32 bit in size (e.g. `int16_t`) will be processed correctly.

I will also correct these in my PR and will make 2 commits for it.

TD-er, 2019-07-28:
Another one....
The pressure is decoded in the packed decoder JS code by dividing it by 10.
But when packing it, it is not multiplied by 10.
This is only done with `#elif ((PAYLOAD_ENCODER == 3) || (PAYLOAD_ENCODER == 4))`
Not with the others.
The plain decoder does not divide by 10.

So this is a mix-up of a lot of decoder/encoder stuff and is either wrong now or will be a breaking change.

TD-er, 2019-07-28:
And another one which is a bit strange.
The `float` representation which is only used in the BME temperature, does write the bytes in the opposite order of what all other values are being stored.
It also does explicit conversion to represent the 2-complement representation, but I think (not entirely sure, should look into it) that a `reinterpret_cast<uint16_t>` should also do the same. (of an int16_t value, not the float value)

cyberman54, 2019-07-28:
The encoder logic has grown over time, using code snippets from different examples and projects. So would be great if you find some time to do a refactoring PR. I think it needs to be a breaking change, then.

TD-er, 2019-07-28:
Is there a reason why the byte order of the 'plain encoded' data is the opposite from the 'packed encoded' data?

cyberman54, 2019-07-28:
Not really i guess. The plain decoder came first, the packed decoder was added later with the goal to have a more efficient payload. Maybe we can drop the plain encoder completely now.

cyberman54, 2019-07-28:
Pressure: I think it's correct, because Cayenne uses 0.1 Hectopascal per bit, while plain/packed encoder just transmits pascal. So we have a factor 10 = 0,1 * 100 difference.

TD-er, 2019-07-28:
So then it's up-to you to decide on:
- Remove plain encoder?
- How to manage "existing nodes" out there?

For example you may choose to force users to create a new TTN application using the 'new' decoder if they either used the plain or packed encoder/decoder.

Then you also need to make a clear schema of what values are represented how. (e.g. `uint16_t pressure` in 0.1 mbar resolution)
The current packed decoder does assume 0.1 mbar resolution:
```js

var pressure = function (bytes) {
    if (bytes.length !== pressure.BYTES) {
        throw new Error('Pressure must have exactly 2 bytes');
    }

    var h = bytesToInt(bytes);
    return +(h / 10).toFixed(1);
};
pressure.BYTES = 2;
```

cyberman54, 2019-07-28:
Maybe best way would be to 
- make fixes in the current encoders, as long as this ist not a breaking change
- create an additional new `PAYLOAD_ENCODER == 5` which fixes all bugs and optimizes payload
- in readme.md state that encoder 5 should be used for new applications

TD-er, 2019-07-28:
So then increasing altitude resolution is better left out for now?

cyberman54, 2019-07-28:
I would say leave it out in legacy decoders, and add it in new decoder.

TD-er, 2019-07-28:
See PR to your development branch: https://github.com/cyberman54/ESP32-Paxcounter/pull/405

cyberman54, 2019-08-18:
Merged.

closed 2019-08-18 by cyberman54