cyberman54/ESP32-Paxcounter #407: T-BEAM + BM680

oscarmh, 2019-08-05:
First of all thanks for Paxcounter, Ive it running on a T-BEAM running perfectly with a small display that is connected 3.3v/Ground/21SCL/22SDA I also want to attach a BM680, can it be done ? Thanks in advance

cyberman54, 2019-08-05:
Just attach the BNA680 to the I2C bus, by wiring it to SDA/SCL, and add BME to the hal config file.
Example for T-Beam:

```
// clang-format off
// upload_speed 921600
// board ttgo-t-beam

#ifndef _TTGOBEAM_H
#define _TTGOBEAM_H

#include <stdint.h>

// Hardware related definitions for TTGO T-Beam board
// (only) for older T-Beam version T22_V05 eternal wiring LORA_IO1 to GPIO33 is needed!
//
// pinouts taken from http://tinymicros.com/wiki/TTGO_T-Beam

#define HAS_LED GPIO_NUM_14 // on board green LED, only new version TTGO-BEAM V07
//#define HAS_LED GPIO_NUM_21 // on board green LED, only old verison TTGO-BEAM V05

#define HAS_LORA 1       // comment out if device shall not send data via LoRa
#define CFG_sx1276_radio 1 // HPD13A LoRa SoC
#define BOARD_HAS_PSRAM // use extra 4MB external RAM
#define HAS_BUTTON GPIO_NUM_39 // on board button (next to reset)
#define BAT_MEASURE_ADC ADC1_GPIO35_CHANNEL // battery probe GPIO pin -> ADC1_CHANNEL_7
#define BAT_VOLTAGE_DIVIDER 2 // voltage divider 100k/100k on board

// GPS settings
#define HAS_GPS 1 // use on board GPS
#define GPS_SERIAL 9600, SERIAL_8N1, GPIO_NUM_12, GPIO_NUM_15 // UBlox NEO 6M
//#define GPS_INT GPIO_NUM_34 // 30ns accurary timepulse, to be external wired on pcb: NEO 6M Pin#3 -> GPIO34

// enable only if device has these sensors, otherwise comment these lines
// BME680 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME680 SDA, SCL
#define BME680_ADDR BME680_I2C_ADDR_PRIMARY // !! connect SDIO of BME680 to GND !!

// display (if connected)
#define HAS_DISPLAY U8X8_SSD1306_128X64_NONAME_HW_I2C
#define MY_OLED_SDA SDA
#define MY_OLED_SCL SCL
#define MY_OLED_RST U8X8_PIN_NONE
//#define DISPLAY_FLIP  1 // use if display is rotated

// user defined sensors (if connected)
//#define HAS_SENSORS 1 // comment out if device has user defined sensors

//#define DISABLE_BROWNOUT 1 // comment out if you want to keep brownout feature

#endif

```

closed 2019-08-05 by cyberman54

oscarmh, 2019-08-05:
thanks a lot for your quick answer.

oscarmh, 2019-08-06:
Sorry to jump  in again, Ive soldered SDA/SCL and GND / 3.3V but even doing the changes in the ttgobeam.h I cannot see the BME data, any hint ? In your comments I see SDIO to GND but no SDIO pin in the BME... thanks in advance

cyberman54, 2019-08-06:
every BME680 has SDIO and it should not left floating according to Bosch datasheet, since i2c adress is undefined then. But you can try to swap i2c adress 0x76 <-> 0x77 in the hal file and try again.

cyberman54, 2019-08-06:
and remember to use clean function in platformio before rebuilding code, without this changes in the hal file may not be compiled.

oscarmh, 2019-08-07:
Must be a new-new BME680 no SDIO but SDL/SDC and 3.3v+GNS, it is working now, pretty well !! Thanks 
[image]

cyberman54, 2019-08-07:
BME680 always has SDIO, but it maybe fixed wired on the PCB to GND or VDD.

oscarmh, 2019-08-07:
for sure, next step is to provide power using a Solar Panel I will work on that, thanks for your support.