cyberman54/ESP32-Paxcounter #409: Kein OLED nur grüne LED

agonaut, 2019-08-12:
Ich habe die Anweisungen aus dem Make Heft 1/2019 befolgt und meinen TTGO T3_V1.6 (ttgov21new.h) geflasht. Leider blieb das OLED dunkel und die RGB LED blinkt schnell grün. Im Artikel und im Youtube von Sebastian Müller ist mir lediglich aufgefallen, dass es sich da um eine ältere Version vom 'ESP32-Paxcounter-master' handeln muss. Die Boards werden anders angeboten 'env_default = ttgov21new.h' anstelle von 'halfile = ttgov21new.h' und nach dem Clean erscheint.
Environment ota [IGNORED]
Environment usb [SUCCESS]
Environment dev [IGNORED]
Sollte ja kein Problem sein?!
Auch nach meiner Fehlersuche mit Hilfe des Tutorials aus den GitHub 'cyberman54/ESP32-Paxcounter' hatte ich keinen Erfolg.
Was mache ich nur falsch?

I followed the instructions in Make Issue 1/2019 and flashed my TTGO T3_V1.6 (ttgov21new.h). Unfortunately, the OLED remained dark and the RGB LED flashes green quickly. In the article and in the Youtube of Sebastian Mueller I only noticed, that this must be an older version of the 'ESP32-Paxcounter-master'. The boards are offered differently 'env_default = ttgov21new.h' instead of 'halfile = ttgov21new.h' and after the clean appears.
Environment ota [IGNORED]
Environment usb [SUCCESS]
Environment dev [IGNORED]
Should not be a problem?
Even after my troubleshooting using the tutorial from the GitHub 'cyberman54 / ESP32 Paxcounter' I had no success.
What am I doing wrong?

Gruß und greeting,
agonaut

cyberman54, 2019-08-13:
Which platformio version did you use?

agonaut, 2019-08-13:
It is version 1.8.2

Visual Studio Code 1.37.0
the latest ESP32 Paxcounter
win10 or macOS High Sierra

cyberman54, 2019-08-13:
This is not Platformio version, should be v4.0, maybe you're still using oldrr 3.6.x, this won't work with current paxcounter code. Please check zur platformio version.

agonaut, 2019-08-13:
Yes it is the Platformio version 4.0. ... the other one was the IDE

cyberman54, 2019-08-13:
@agonaut 
Did you comment the line
`;halfile = generic.h`
in platformio.ini, when uncommenting the line
`halfile = ttgov21new.h`
?

If not, wrong hardware profile "generic" was build. The buildscript is using the first halfile which is found in platformio.ini.

If this was not the problem, please write serial log when running the device (compile with debug level = 4) and post the log here.

agonaut, 2019-08-14:
yes, is set correctly (halfile = ttgov21new.h)
I do not know how to use the debugger properly
[image]
[debug.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/3499925/debug.txt)

cyberman54, 2019-08-14:
Don't use the debugger, just write serial console log and post it here. Without giving further information no one will be able to help you.

agonaut, 2019-08-14:
I'm sorry, I'm the total beginner.

[build.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/3501551/build.txt)
[upload.txt](https://github.com/cyberman54/ESP32-Paxcounter/files/3501552/upload.txt)

cyberman54, 2019-08-14:
We need a serial log of the device's output, not a screenshot of platformio output. Open serial terminal in platformio while device is connected.

This is a not a beginners support forum. Maybe you get help from a beginner's forum or start with some simple "hello world" projects to get used to your toolchain, before opening issues?

agonaut, 2019-08-15:
> Executing task: platformio.exe device monitor <

--- Miniterm on COM3  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[I][main.cpp:348] setup(): Starting Wifi...
I (4937) wifi: wifi driver task: 3ffc384c, prio:23, stack:3584, core=0
I (10056) wifi: wifi firmware version: 6d404d4
I (10058) wifi: config NVS flash: disabled
I (10062) wifi: config nano formating: disabled
I (10067) wifi: Init dynamic tx buffer num: 32
I (10070) wifi: Init data frame dynamic rx buffer num: 10
I (10075) wifi: Init management frame dynamic rx buffer num: 10       
I (10081) wifi: Init static rx buffer size: 1600
I (10085) wifi: Init static rx buffer num: 8
I (10089) wifi: Init dynamic rx buffer num: 10
I (10094) wifi: set country: cc=EU␀ schan=1 nchan=13 policy=1

W (10099) wifi: ap start fail

I (10158) wifi: ic_enable_sniffer
[I][main.cpp:362] setup(): Starting Interrupt Handler...
[I][main.cpp:392] setup(): Starting Timers...
[I][main.cpp:453] setup(): Starting Timekeeper...
[I][timekeeper.cpp:109] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][main.cpp:460] setup(): Features: LED BATT OTA LORA FILTER OLED PACKED WIFI      
[I][lorawan.cpp:339] onEvent(): JOINING
[I][lorawan.cpp:339] onEvent(): JOIN WAIT
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 4010mV
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 4016mV
[I][senddata.cpp:73] sendCounter(): Counter cleared

cyberman54, 2019-08-15:
Looks good so far. Set debug level = 4 to get deeper logging.

Did you check that the display hardware is working, with other program?

closed 2019-08-18 by cyberman54