cyberman54/ESP32-Paxcounter #414: T-BEAM 1.7.93 Activation in TTN no uplink

oscarmh, 2019-08-24:
Hi, I´m struggling on having my T-Beam talking to TTN, any help ?
Clean git clone to platformio project
1) Configured platformio.ini to use the halfile = ttgobeam.h
2) loraconf.h according data from ttn
3) ttgobeam.h added BME680 
4) DRAGINO TG02 in place SF=7 according to paxcounter.conf and configured to listen 868.100000

I got activation from TTN but no uplink... any suggestion ?

I added DEBUG in lmic and paxcoutner.conf and got

20977767: TXMODE, freq=868100000, len=23, SF=8, BW=125, CR=4/5, IH=0
[D][cyclic.cpp:29] doHousekeeping(): IRQhandler 2196 bytes left | Taskstate = 0
[D][cyclic.cpp:32] doHousekeeping(): LMiCtask 2340 bytes left | Taskstate = 2
[D][cyclic.cpp:36] doHousekeeping(): Gpsloop 1460 bytes left | Taskstate = 1
[D][cyclic.cpp:51] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
[I][cyclic.cpp:57] doHousekeeping(): Voltage: 4310mV
[I][cyclic.cpp:64] doHousekeeping(): BME680 Temp: 30.78°C | IAQ: 39.76 | IAQacc: 1
start single rx: now-rxtime: 9
21281634: RXMODE_SINGLE, freq=868100000, SF=8, BW=125, CR=4/5, IH=0
rxtimeout: entry: 21314356 rxtime: 21281597 entry-rxtime: 32759 now-entry: 6 rxtime-txend: 296692
start single rx: now-rxtime: 37

oscarmh, 2019-08-24:
I´m using ttgobeam.h like that 
// clang-format off
// upload_speed 921600
// board ttgo-t-beam

#ifndef _TTGOBEAM_H
#define _TTGOBEAM_H

#include <stdint.h>

// Hardware related definitions for TTGO T-Beam board
// (only) for older T-Beam version T22_V05 eternal wiring LORA_IO1 to GPIO33 is needed!
//
// pinouts taken from http://tinymicros.com/wiki/TTGO_T-Beam

#define HAS_LED GPIO_NUM_14 // on board green LED, only new version TTGO-BEAM V07
//#define HAS_LED GPIO_NUM_21 // on board green LED, only old verison TTGO-BEAM V05

#define HAS_LORA 1       // comment out if device shall not send data via LoRa
#define CFG_sx1276_radio 1 // HPD13A LoRa SoC
#define BOARD_HAS_PSRAM // use extra 4MB external RAM
#define HAS_BUTTON GPIO_NUM_39 // on board button (next to reset)
#define BAT_MEASURE_ADC ADC1_GPIO35_CHANNEL // battery probe GPIO pin -> ADC1_CHANNEL_7
#define BAT_VOLTAGE_DIVIDER 2 // voltage divider 100k/100k on board

// GPS settings
#define HAS_GPS 1 // use on board GPS
#define GPS_SERIAL 9600, SERIAL_8N1, GPIO_NUM_12, GPIO_NUM_15 // UBlox NEO 6M
//#define GPS_INT GPIO_NUM_34 // 30ns accurary timepulse, to be external wired on pcb: NEO 6M Pin#3 -> GPIO34

// enable only if device has these sensors, otherwise comment these lines
// BME680 sensor on I2C bus
#define HAS_BME 1 // Enable BME sensors in general
#define HAS_BME680 SDA, SCL
#define BME680_ADDR 0x76 // !! connect SDIO of BME680 to GND !!

// display (if connected)
#define HAS_DISPLAY U8X8_SSD1306_128X64_NONAME_HW_I2C
#define MY_OLED_SDA SDA
#define MY_OLED_SCL SCL
#define MY_OLED_RST U8X8_PIN_NONE
//#define DISPLAY_FLIP  1 // use if display is rotated

// user defined sensors (if connected)
//#define HAS_SENSORS 1 // comment out if device has user defined sensors

//#define DISABLE_BROWNOUT 1 // comment out if you want to keep brownout feature

#endif

oscarmh, 2019-08-24:
I´ve made progress, I got a new T-BEAM never flashed with PAXCOUNTER and it works !! COuld it be related to NVRAM settings ? How can I force eraseConfig() each time it boots ??

cyberman54, 2019-08-24:
Dragino Gateway is not LoRaWAN compliant. So this can't work.

oscarmh, 2019-08-24:
[image]
it is !!

cyberman54, 2019-08-24:
Dragino ist a dual channel gateway and not LoRaWAN compliant, so won't work reliable with LMiC. Please retry with a full compliant gateway. If problem persists you may reopen this issue.

closed 2019-08-24 by cyberman54