cyberman54/ESP32-Paxcounter #419: TTGO v1 + BME680 ?

oscarmh, 2019-08-31:
Hi, I´m trying to have the BME680 running on the TTGOv1, but I´m not able to make it run, it recognizes de BME but values are always zero, any hint ? For sure you have a reason it is not implemented but Ive no found any on that in the issues section, thanks in advance

cyberman54, 2019-08-31:
Please give a more precise description of your problem / issue here.

Didn't you post an issue regarding a non Lorawan compliant Dragino gateway? This won't work with the LMiC stack used in paxcounter.

oscarmh, 2019-08-31:
Hi thanks, I managed to install a full-LORA compliant LG-308 my only issue is to make the BME608 work with the small TTGO v01 the one that does not have SD reader nor GPS, I tried to add the BMC definition to the hal file but I only get zeros as the values returned by BME608

I][cyclic.cpp:66] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[I][cyclic.cpp:66] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0

when i boot the TTGO I got

[I][main.cpp:131] setup(): Starting PAXCNT v1.7.974
[I][main.cpp:144] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
[I][main.cpp:146] setup(): Internal Total heap 264132, internal Free Heap 238136
[I][main.cpp:152] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2-18-g977854975
[I][main.cpp:154] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:155] setup(): Wifi/BT software coexist version 1.1.5
[I][main.cpp:159] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:164] setup(): Arduino LMIC version 2.3.2.71

on the features 

I][timekeeper.cpp:150] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][main.cpp:451] setup(): Features: OLED LED OTA LORA FILTER PACKED WIFI BME680 BTN_PD

thanks in advance for your help

cyberman54, 2019-08-31:
Check your payloadmask (is displayed during startup), what value is displayed?

Try to switch on BME data by sending command 15 01 on Port 2.

cyberman54, 2019-08-31:
I found a bug in new feature payload masking. Fixed it. Please retry current v1.7.979 from master, and report here if this solves your issue, thanks.

oscarmh, 2019-09-01:
hi !! testing the new v1.7.979

NVRAM cleared. activated BME on hal file for ttgov1.h

**Boot process** 

I][main.cpp:131] setup(): Starting PAXCNT v1.7.979
[I][main.cpp:144] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 4MB external Flash
[I][main.cpp:146] setup(): Internal Total heap 263732, internal Free Heap 237736
[I][main.cpp:152] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2-18-g977854975
[I][main.cpp:154] setup(): Flash Size 4194304, Flash Speed 40000000
[I][main.cpp:155] setup(): Wifi/BT software coexist version 1.1.5
[I][main.cpp:159] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:164] setup(): Arduino LMIC version 2.3.2.71
[I][macsniff.cpp:36] printKey(): DevEUI: CC50E3B608C4FEEC
[I][macsniff.cpp:36] printKey(): AppEUI: <APPEUI>
[I][macsniff.cpp:36] printKey(): AppKey: <APPEUI>
[I][configmanager.cpp:189] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:54] open_storage(): Opening NVS
[I][configmanager.cpp:59] open_storage(): Done
[I][configmanager.cpp:203] loadConfig(): NVRAM settings version = 1.7.979

**1501 upload con port 2**
hal file explicit BME680 SDA and SCL for pins 4,15

**Connection to TTN done **
[D][cyclic.cpp:29] doHousekeeping(): IRQhandler 2376 bytes left | Taskstate = 0
[D][cyclic.cpp:32] doHousekeeping(): LMiCtask 2308 bytes left | Taskstate = 2
[D][cyclic.cpp:34] doHousekeeping(): Lorasendtask 1304 bytes left | Taskstate = 2
[D][cyclic.cpp:53] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
[I][cyclic.cpp:66] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[D][cyclic.cpp:29] doHousekeeping(): IRQhandler 2376 bytes left | Taskstate = 0
[D][cyclic.cpp:32] doHousekeeping(): LMiCtask 2308 bytes left | Taskstate = 1
[D][cyclic.cpp:34] doHousekeeping(): Lorasendtask 1304 bytes left | Taskstate = 2
[D][cyclic.cpp:53] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
[I][cyclic.cpp:66] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[I][senddata.cpp:75] sendData(): Counter cleared
[I][lorawan.cpp:299] onEvent(): TX START
[I][lorawan.cpp:378] lora_send(): 2 byte(s) sent to LORA
[I][lorawan.cpp:299] onEvent(): TX COMPLETE
[I][lorawan.cpp:299] onEvent(): TX START
[I][lorawan.cpp:378] lora_send(): 8 byte(s) sent to LORA
[I][lorawan.cpp:299] onEvent(): TX COMPLETE

Wifis detected ok
Display as usual
UPLOAD always payload 0000000000000000

Any hint ?

oscarmh, 2019-09-01:
adding extra info sent 1501

[I][lorawan.cpp:554] myRxCallback(): Received 2 bytes of payload on port 2
[I][lorawan.cpp:563] myRxCallback(): Received 13 MAC messages:
[I][lorawan.cpp:651] mac_decode(): Unknown MAC message 0xA0
[I][lorawan.cpp:651] mac_decode(): Unknown MAC message 0xE6
[I][lorawan.cpp:651] mac_decode(): Unknown MAC message 0x27
[I][lorawan.cpp:643] mac_decode(): Network command ResetConf
[I][lorawan.cpp:651] mac_decode(): Unknown MAC message 0x85
[I][lorawan.cpp:643] mac_decode(): Network command ResetConf
[I][lorawan.cpp:643] mac_decode(): Network command LinkADRReq
[I][rcommand.cpp:135] set_bme(): Remote command: set BME mode to on
[I][configmanager.cpp:78] saveConfig(): Storing settings in NVS
[I][configmanager.cpp:54] open_storage(): Opening NVS
[I][configmanager.cpp:59] open_storage(): Done
[I][configmanager.cpp:170] saveConfig(): Done
[D][cyclic.cpp:29] doHousekeeping(): IRQhandler 2376 bytes left | Taskstate = 0
[D][cyclic.cpp:32] doHousekeeping(): LMiCtask 2260 bytes left | Taskstate = 1
[D][cyclic.cpp:34] doHousekeeping(): Lorasendtask 1256 bytes left | Taskstate = 2
[D][cyclic.cpp:53] doHousekeeping(): LEDloop 484 bytes left | Taskstate = 2
[I][cyclic.cpp:66] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[I][lorawan.cpp:299] onEvent(): TX START
[I][lorawan.cpp:299] onEvent(): TX COMPLETE

cyberman54, 2019-09-07:
This was caused by a wrong user set payloadmask. No issue. Closed.

closed 2019-09-07 by cyberman54