cyberman54/ESP32-Paxcounter #422: Please add TTGO T-Beam v1.0 SX192 support.

Habies, 2019-09-02:
The GPS U-blox wil not start, cannot get a gps fix.

cyberman54, 2019-09-02:
I don't own a TTGO v1.0 for testing.
If i make a quick & dirty coding in development branch, can you test it and report results here?

Habies, 2019-09-02:
Yes, i can test and report back if you do the code.

cyberman54, 2019-09-02:
@Habies Done. Please test current code in development branch and report here. Thanks.

Habies, 2019-09-02:
@cyberman54 Thanks for the fast support, the file i2scan.cpp has an error.
expected declaration '}' token

cyberman54, 2019-09-02:
@Habies  fixed. And i created a separate `ttgobeam10.h` file with setting for TTGO Beam V10.
Please use it and re-test, thanks.

Habies, 2019-09-02:
@cyberman54 Still no luck with the GPS. The APX192 chip is detected and the GPX_RX and TX pins are correct.
[I][main.cpp:169] setup(): TinyGPS+ version 1.0.2
[I][i2cscan.cpp:20] i2c_scan(): Starting I2C bus scan...
[I][i2cscan.cpp:46] i2c_scan(): 0x34: AXP192 power management
[I][i2cscan.cpp:99] AXP192_init(): AXP192 PMU initialized.
[I][i2cscan.cpp:37] i2c_scan(): 0x3C: SSD1306 Display controller
[I][i2cscan.cpp:63] i2c_scan(): I2C scan done, 2 devices found.
[I][configmanager.cpp:189] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:54] open_storage(): Opening NVS
[I][configmanager.cpp:59] open_storage(): Done
[I][configmanager.cpp:203] loadConfig(): NVRAM settings version = 1.8.01
[I][configmanager.cpp:221] loadConfig(): bsecstate = 0
[I][configmanager.cpp:226] loadConfig(): lorasf = 9
[I][configmanager.cpp:234] loadConfig(): txpower = 15
[I][configmanager.cpp:242] loadConfig(): adrmode = 1
[I][configmanager.cpp:250] loadConfig(): screensaver = 0
[I][configmanager.cpp:258] loadConfig(): screenon = 1
[I][configmanager.cpp:266] loadConfig(): countermode = 0
[I][configmanager.cpp:274] loadConfig(): sendcycle = 30
[I][configmanager.cpp:282] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:290] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:298] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:306] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:314] loadConfig(): blescantime = 8
[I][configmanager.cpp:322] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:330] loadConfig(): rssilimit = 0
[I][configmanager.cpp:338] loadConfig(): payloadmask = 127
[I][configmanager.cpp:346] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:354] loadConfig(): Run mode = 0
[I][configmanager.cpp:361] loadConfig(): Done
[I][main.cpp:188] setup(): PSRAM found and initialized
[I][gpsread.cpp:37] gps_init(): Using serial GPS
[I][main.cpp:284] setup(): Starting GPS Feed...
[I][lorawan.cpp:422] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:425] lora_stack_init(): Starting LMIC...
[I][main.cpp:352] setup(): Starting Wifi...
I (6374) wifi: wifi driver task: 3ffb21c4, prio:23, stack:3584, core=0
[I][lorawan.cpp:316] onEvent(): JOINING
I (12959) wifi: wifi firmware version: 6d404d4
I (12964) wifi: config NVS flash: disabled
I (12968) wifi: config nano formating: disabled
I (12973) wifi: Init dynamic tx buffer num: 32
I (12976) wifi: Init data frame dynamic rx buffer num: 10
I (12981) wifi: Init management frame dynamic rx buffer num: 10
I (12987) wifi: Init static rx buffer size: 1600
I (12992) wifi: Init static rx buffer num: 8
I (12995) wifi: Init dynamic rx buffer num: 10
I (13000) wifi: set country: cc=EU

Habies, 2019-09-02:
@cyberman54 Sorry, spoke to soon, GPS data is working! Great job! Now trying to get the battery monitoring back and add some sensors.

cyberman54, 2019-09-02:
Battery monitoring should aready be back, i did implement it, too.

cyberman54, 2019-09-02:
But you need to enable it now with rcommand 16 01 on Port 2, or change the defaults in configmanager.cpp

Habies, 2019-09-02:
Ok, thanks!

cyberman54, 2019-09-02:
I will merge code to Master and close this issue. But you may keep noting here any hints.

closed 2019-09-02 by cyberman54