cyberman54/ESP32-Paxcounter #428: Error - UnconfirmedDataUp - Loraserver.io

viniaero, 2019-09-07:
Hello,
I can not receive the data, I set 3 = Cayenne LPP dynamic. but it doesn't work

[image]

[image]

[image]

What can I do wrong?

cyberman54, 2019-09-07:
Probably you have issues with your setup. Try TTN.

viniaero, 2019-09-07:
I just tested on TTN, is it working, what can it be?

cyberman54, 2019-09-07:
This is not a Loraserver issue tracker. Please ask on a Loraserver forum, thanks.

closed 2019-09-07 by cyberman54