cyberman54/ESP32-Paxcounter #437: HELTEC LORA 32 V2 - OTAA join fails - Paxcounter 1.8.02

FrankUlbrich, 2019-09-10:
Hi  cyberman,
I got my problem back with my HELTEC LORA 32 v2 device and I got the feeling, that this hardware is not good for the paxcounter.

https://github.com/cyberman54/ESP32-Paxcounter/issues/411

The join of PAXCOUNTER does not working in my case.
In the file of Loraserver the request is recorded and the positive reponse is sent.
It looks like that this response is not recognized by the node.

I modified the setting but it does not solve the problem.
The old 1.7 code does suddenly have problem, too.: 
// ATTN: VALUES > 7 WILL CAUSE RECEPTION AND JOIN PROBLEMS WITH HIGH SF RATES
#define CLOCK_ERROR_PROCENTAGE 7 

### PAXCNT v1.8.02

```
[I][main.cpp:131] setup(): Starting PAXCNT v1.8.02
[I][main.cpp:144] setup(): This is ESP32 chip with 2 CPU cores, WiFi/BT/BLE, silicon revision 1, 8MB external Flash
[I][main.cpp:146] setup(): Internal Total heap 265984, internal Free Heap 239740
[I][main.cpp:152] setup(): ChipRevision 1, Cpu Freq 240, SDK Version v3.2-18-g977854975
[I][main.cpp:154] setup(): Flash Size 8388608, Flash Speed 40000000
[I][main.cpp:155] setup(): Wifi/BT software coexist version 1.1.5
[I][main.cpp:159] setup(): IBM LMIC version 1.6.1468577746
[I][main.cpp:164] setup(): Arduino LMIC version 3.0.99.3
[I][macsniff.cpp:36] printKey(): DevEUI: 00EC094D04299F02
[I][macsniff.cpp:36] printKey(): AppEUI: 70B3D57ED0020457
[I][macsniff.cpp:36] printKey(): AppKey: C4154AE996751303B6691E0D43AF65DA
[W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(4)=0, scl(15)=0
[E][esp32-hal-i2c.c:1426] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=0, scl=0
[I][i2cscan.cpp:13] i2c_scan(): Starting I2C bus scan...
[I][i2cscan.cpp:59] i2c_scan(): I2C scan done, 0 devices found.
[I][configmanager.cpp:189] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:54] open_storage(): Opening NVS
[I][configmanager.cpp:59] open_storage(): Done
[I][configmanager.cpp:203] loadConfig(): NVRAM settings version = 1.8.02
[I][configmanager.cpp:221] loadConfig(): bsecstate = 0
[I][configmanager.cpp:226] loadConfig(): lorasf = 9
[I][configmanager.cpp:234] loadConfig(): txpower = 15
[I][configmanager.cpp:242] loadConfig(): adrmode = 1
[I][configmanager.cpp:250] loadConfig(): screensaver = 0
[I][configmanager.cpp:258] loadConfig(): screenon = 1
[I][configmanager.cpp:266] loadConfig(): countermode = 0
[I][configmanager.cpp:274] loadConfig(): sendcycle = 30
[I][configmanager.cpp:282] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:290] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:298] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:306] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:314] loadConfig(): blescantime = 8
[I][configmanager.cpp:322] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:330] loadConfig(): rssilimit = 0
[I][configmanager.cpp:338] loadConfig(): payloadmask = 127
[I][configmanager.cpp:346] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:354] loadConfig(): Run mode = 0
[I][configmanager.cpp:361] loadConfig(): Done
[I][main.cpp:241] setup(): Starting LED Controller...
[I][power.cpp:153] calibrate_voltage(): ADC characterization based on reference voltage stored in eFuse
[I][lorawan.cpp:432] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:435] lora_stack_init(): Starting LMIC...
RXMODE_RSSI
[I][main.cpp:364] setup(): Starting Wifi...
I (4988) wifi: wifi driver task: 3ffb973c, prio:23, stack:3584, core=0
I (10151) wifi: wifi firmware [I][lorawan.cpp:322] onEvent(): JOINING
version: 6d404d4
312207: engineUpdate, opmode=0x4
I (10156) wifi: config NVS flash: disabled
I (10163) wifi: config nano formating: disabled
I (10168) wifi: Init dynamic tx buffer num: 32
I (10171) wifi: Init data frame dynamic rx buffer num: 10
I (10177) wifi: Init management frame dynamic rx buffer num: 10
I (10182) wifi: Init static rx buffer size: 1600
I (10187) wifi: Init static rx buffer num: 8
I (10190) wifi: Init dynamic rx buffer num: 10
I (10195) wifi: set country: cc=EU␀ schan=1 nchan=13 policy=1

W (10200) wifi: ap start fail

I (10203) wifi: Set ps type: 0

I (10261) wifi: ic_enable_sniffer
[I][main.cpp:378] setup(): Starting Interrupt Handler...
[I][main.cpp:400] setup(): Starting Timers...
[I][main.cpp:458] setup(): Starting Timekeeper...
[I][timekeeper.cpp:150] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
[I][main.cpp:465] setup(): Features: VEXT OLED LED BATT OTA LORA FILTER PACKED WIFI BTN_PD
512838: engineUpdate, opmode=0x4
512988: TXMODE, freq=868500000, len=23, SF=7, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 8
806908: RXMODE_SINGLE, freq=868500000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 823236 rxtime: 806889 entry-rxtime: 16347 now-entry: 52 rxtime-txend: 290036
start single rx: now-rxtime: 3
867408: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 930903 rxtime: 867395 entry-rxtime: 63508 now-entry: 10 rxtime-txend: 350542
[I][lorawan.cpp:322] onEvent(): JOIN WAIT
931143: engineUpdate, opmode=0x4
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3848mV
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3524mV
[I][senddata.cpp:76] sendData(): Counter cleared
4922149: engineUpdate, opmode=0x4
4922315: TXMODE, freq=868100000, len=23, SF=7, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 50
5216277: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 5232604 rxtime: 5216215 entry-rxtime: 16389 now-entry: 51 rxtime-txend: 290036
start single rx: now-rxtime: 4
5276740: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 5340230 rxtime: 5276721 entry-rxtime: 63509 now-entry: 128 rxtime-txend: 350542
[I][lorawan.cpp:322] onEvent(): JOIN WAIT
5340580: engineUpdate, opmode=0x4
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3848mV
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3826mV
[I][senddata.cpp:76] sendData(): Counter cleared
8985086: engineUpdate, opmode=0x4
8985233: TXMODE, freq=868300000, len=23, SF=7, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 9
9279153: RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=0
rxtimeout: entry: 9295482 rxtime: 9279134 entry-rxtime: 16348 now-entry: 42 rxtime-txend: 290036
start single rx: now-rxtime: 4
9339654: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 9403149 rxtime: 9339640 entry-rxtime: 63509 now-entry: 143 rxtime-txend: 350542
[I][lorawan.cpp:322] onEvent(): JOIN WAIT
9403528: engineUpdate, opmode=0x4
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3612mV
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3636mV
[I][senddata.cpp:76] sendData(): Counter cleared
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3706mV
13849283: engineUpdate, opmode=0x4
13849412: TXMODE, freq=868500000, len=23, SF=8, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 3
14146662: RXMODE_SINGLE, freq=868500000, SF=8, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 9
14207055: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 14270543 rxtime: 14207035 entry-rxtime: 63508 now-entry: 106 rxtime-txend: 350542
[I][lorawan.cpp:322] onEvent(): JOIN WAIT
14270893: engineUpdate, opmode=0x4
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3684mV
[I][senddata.cpp:76] sendData(): Counter cleared
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3558mV
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3630mV
[I][senddata.cpp:76] sendData(): Counter cleared
[I][cyclic.cpp:59] doHousekeeping(): Voltage: 3660mV
21480772: engineUpdate, opmode=0x4
21480902: TXMODE, freq=868100000, len=23, SF=8, BW=125, CR=4/5, IH=0
start single rx: now-rxtime: 4
21778152: RXMODE_SINGLE, freq=868100000, SF=8, BW=125, CR=4/5, IH=0
rxtimeout: entry: 21810800 rxtime: 21778138 entry-rxtime: 32662 now-entry: 105 rxtime-txend: 290155
start single rx: now-rxtime: 54
21838590: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
rxtimeout: entry: 21902085 rxtime: 21838525 entry-rxtime: 63560 now-entry: 7 rxtime-txend: 350542
[I][lorawan.cpp:322] onEvent(): JOIN WAIT
21902331: engineUpdate, opmode=0x4
```

OTAA won't work with my device. Demo code does not have a prblem.
On the other hand. my "TTGO BEAM" works fine.

I would like to let you know, that I love your project.

FrankUlbrich, 2019-09-10:
[image]

The accept come back from the gateway

cyberman54, 2019-09-11:
This is probably caused by recent changes in LMIC library, since the author of this library is working on the goal to reach certification.

You can set back to former stable LMIC version 2.3.2 by editing platformio.ini, i think this will solve the problem, until next point release of LMIC stack is out.

FrankUlbrich, 2019-09-11:
Thank you very much for your answer and spending time into my issue. I will try all your coming releases and combinations and report if I get success.

cyberman54, 2019-09-11:
Does your Heltev V2 never join? Or does is join after a few minutes, while displaying JOIN WAIT?

cyberman54, 2019-09-17:
Current MCCI LMIC version is working again.