cyberman54/ESP32-Paxcounter #438: Not flashing on tbeam v1.0

netmonk, 2019-09-11:
I just received the latest lilygo tbeam board t22-v1.0 20190612. Seems you merged your sources tree to support this board. 

When using plateformio with Code on linux, i get this message : 
```
Retrieving maximum program size .pio/build/usb/firmware_ttgobeam10_v1.8.02.elf
Building .pio/build/usb/firmware_ttgobeam10_v1.8.02.bin
Checking size .pio/build/usb/firmware_ttgobeam10_v1.8.02.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   5.3% (used 70120 bytes from 1310720 bytes)
PROGRAM: [====Error: The program size (1674946 bytes) is greater than maximum allowed (1310720 bytes)
======]  12*** [checkprogsize] Explicit exit, status 1
7.8% (used 1674946 bytes from 1310720 bytes)
```

I tried to change the board value in the plateformio.ini to `ttgo-t-beam` with board def you can find in `.platformio/platforms/espressif32/boards`
This time it uploads correctly, but screen not working although i declared it in ttgobeam10.h
```
// display (if connected)
#define HAS_DISPLAY U8X8_SSD1306_128X64_NONAME_HW_I2C
#define MY_OLED_SDA 21
#define MY_OLED_SCL 22
#define MY_OLED_RST 5
//#define DISPLAY_FLIP  1 // use if display is rotated
``` 
Several time i had also some error with vbat sensor : 

And i got this error to : 
```
Compiling .pio/build/usb/lib951/MCCI LoRaWAN LMIC library/hal/getpinmap_ttgo_lora32_v1.cpp.o
.pio/libdeps/usb/MCCI LoRaWAN LMIC library/src/hal/getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
.pio/libdeps/usb/MCCI LoRaWAN LMIC library/src/hal/getpinmap_thisboard.cpp:61:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
``` 

Can you help ?

cyberman54, 2019-09-12:
I cannot reproduce this, i see 87,5% memory utilization.
Which platformio version do you have in use?

> Building .pio\build\usb\firmware_ttgobeam10_v1.8.03.bin
> esptool.py v2.6
> Memory Usage -> http://bit.ly/pio-memory-usage
> DATA:    [          ]   4.1% (used 54016 bytes from 1310720 bytes)
> PROGRAM: [========= ]  87.5% (used 1147398 bytes from 1310720 bytes)

> I tried to change the board value in the plateformio.ini to ttgo-t-beam with board def you can find in .platformio/platforms/espressif32/boards
> This time it uploads correctly, but screen not working although i declared it in ttgobeam10.h

This war you are using the original T-Beam pin definitions, served by `pins_arduino.h` from the esp32-arduino library. This won't work, because definitions for paxcounter are missing then.

> #pragma message("Board not supported -- use an explicit pinmap")

This is normal (and intentional by the author of the lib), doesn't affect anything here. You can ignore this.

cyberman54, 2019-09-18:
Fixed.

cyberman54, 2019-09-18:
@netmonk I found a workaround to ensure selection of partition table. Could you please re-test? Thanks.

closed 2019-09-18 by cyberman54

netmonk, 2019-09-19:
i'm checking

netmonk, 2019-09-19:
it works ! i didnt test lora part cause i dont have lora gw currently, but gps/wifi/ble/screen is working nice no more upload issue