cyberman54/ESP32-Paxcounter #465: Timesync precision broke with enhancement of timesync protocol

cyberman54, 2019-10-05:
With enhancement of timesync protocol the precision of lora time sync became bad. Before it was 5-10ms, now it's +/- 500ms. Perhaps the second fractions are no longer processed? To be analyzed further.

cyberman54, 2019-10-05:
This is perhaps a side effect of the new display handling with deferred page refresh. That would mean time _is_ precise, but it's not displayed in realtime. I need to check this with digiscope using the IF482 output function of the software.

mariusgripp, 2019-10-10:
Is this about the "Offset gateway to server" in Node-RED? Because I am getting about 32-36ms, which is still more than before, but also not as high as the offset you are getting.

cyberman54, 2019-10-10:
I need to recheck this with digiscope and testboard. Will need to find some time.
Anyway, any value below 50ms would be satisfying.

mariusgripp, 2019-10-11:
With a small python script, I just calculated the average diff between my computer's clock time and the time on an Heltec after timy sync. This diff is about 30ms, which should be fine.

cyberman54, 2020-03-17:
This problem should now be solved, after a deep cleanup of timesync code.

closed 2020-03-17 by cyberman54