cyberman54/ESP32-Paxcounter #468: Support function turn off over push-button for TTGO T-Beam, new version

theArcher73, 2019-10-11:
The TTGO-Beam in the new version has no power slide switch but only button. In the standard version (script of the delivery) a function is implemented which can be used to switch off the beam. If it is possible to integrate this function into the Sketch of the Paxcounter, always wait until the battery is empty, .... - not so nice. 

Thanks a lot!
Archer

cyberman54, 2019-10-11:
This "on/off button longpress" feature is already implemented in current code in development branch, and will be part of master with next release. Please use (and test) code in development branch with your T-Beam V10. Long press of left most button = power off (sleep mode) / short press of middle button = wake up (after sleep).

cyberman54, 2019-10-11:
@theArcher73 oh wait, even better: The function is already part of master code, since v1.8.34.
So just rebuild from current master, reflash, and you have it.

closed 2019-10-11 by cyberman54