cyberman54/ESP32-Paxcounter #471: ttgo v1 board cant join ttn

raystream, 2019-10-16:
Hello,
i have a ttgo v1 board here and flashed the pax counter 1.9.6 on it.
I changed the config files as described but i only get a JOIN_WAIT.
And i can not see any Data in ttn.

here is the log output:

`I (21237) wifi: ic_enable_sniffer
[I][main.cpp:379] setup(): Starting Interrupt Handler...
[I][main.cpp:401] setup(): Starting Timers...
[I][main.cpp:459] setup(): Starting Timekeeper...
[I][timekeeper.cpp:150] timepulse_init(): Timepulse: internal (ESP32 hardware timer)
`

cyberman54, 2019-10-16:
Yes, we have a general join problem.
Unfortunately i could not manage to find the root cause yet.
Any help and analysis is welcome.

cyberman54, 2019-10-17:
Trying to get some help [here](https://www.thethingsnetwork.org/forum/t/need-help-with-mcci-lmic-and-ttn-join-wait-issue/30846).

cyberman54, 2019-10-22:
Fixed. Please reload current version and try again.

closed 2019-10-22 by cyberman54