cyberman54/ESP32-Paxcounter #479: Program size is greater than maximum allowed.

ajpaezm, 2019-10-29:
I'm getting this message for different reasons:

[image]

Some of the scenarios for which this happens are the following:

1) Having the WIFI and BLE sniff flags simultaneously on. 
2) Sometimes by making slight changes to the code (I added a couple of functions for some minor enhancements, like allowing to change the SALT remotely). 

And others, there aren't so many, but they are usually similar to the two described above. I'm constantly going back to the previous version of the code that worked for me but sooner or later I stumble with that again. 

Any hints of how to bypass this issue?

Thanks in advance!

cyberman54, 2019-10-29:
You're using version 1.7.8., this is obsolete. 
Solution is to upgrade code to current version (1.9.71), then this issue should disappear.

The code is not too big, but in a previous version the build script did not select (bigger) partition table `min_spiffs.csv`.

If it works after code update, please close this issue, otherwise report again here.

ajpaezm, 2019-10-31:
It worked! Thanks! :)

closed 2019-10-31 by ajpaezm

vanminh1310, 2021-08-02:
[image]
Hello i am a newbie. Currently I'm getting this error with my program trying to combine the Wifi and BLE libraries on the ESP32. Can anybody show me how to solve it. Thank you very much

cyberman54, 2021-08-02:
@vanminh1310 please post your platformio.ini

vanminh1310, 2021-08-02:
@cyberman54  
[image]

cyberman54, 2021-08-02:
this is from some other project, not paxcounter.

vanminh1310, 2021-08-02:
@cyberman54 
I don't quite understand what you mean.

cyberman54, 2021-08-02:
You have a completely wrong platformio.ini file

vanminh1310, 2021-08-02:
@cyberman54 
Can you help me. I really can't fix it