cyberman54/ESP32-Paxcounter #480: Using LoRa32 V2.1, not joining to LoRaWAN.

ajpaezm, 2019-11-02:
I'm switching boards from the TTGO T-BEAM to the board described in this commentary: https://github.com/LilyGO/TTGO-LORA32-V2.0/issues/10 (the link to the board is this one: https://es.aliexpress.com/item/TTGO-ESP32-Paxcounter-LoRa32-V2-1-la-versi-n-1-6-433-868-915-MHz-LoRa/32915894264.html?spm=a2g0s.12269583.0.0.2d7ba03fPuQKvQ). 

When I make my first build, everything seems to go good, but I never get the message that says:

[I][lorawan.cpp:338] onEvent(): JOINING
[I][lorawan.cpp:223] onEvent(): DEVaddr=xxxxxxxx
[I][lorawan.cpp:338] onEvent(): JOINED

Even though I can see the first payload of counters being sent on the LoRaWAN gateway, the sending of data stops there. I see no further messages. 

On my current setup I'm only reading WiFi, I've enabled the file heltecv2.h (which I think is the correct one) and in that file, besides all the other things coded in them, I added these few lines for a SD Card and an RTC. 

#define HAS_RTC GPIO_NUM_21, GPIO_NUM_22 // SDA, SCL

#define SD_SCK GPIO_NUM_4
#define SD_MISO GPIO_NUM_13
#define SD_MOSI GPIO_NUM_2
#define SD_CS GPIO_NUM_25
#define SD_RST 5

There's a difference between this boards version V2.0 and V2.1 on the LoRa pins that I'm not sure are already reflected in the current setup (is the same as is written in [heltecv2.h](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/hal/heltecv2.h))
[image]

What could be the source of the issue? Has anyone else encountered this before?

Thanks in advance for any help.

cyberman54, 2019-11-02:
Check setting for verbose in paxcounter.conf, and setting for debug_level in platformio.ini.

cyberman54, 2019-11-02:
Select ttgov21new.h in platformio.ini

ajpaezm, 2019-11-02:
Great! Thanks!

closed 2019-11-02 by ajpaezm