cyberman54/ESP32-Paxcounter #482: question sending gps data on port1

nerdyscout, 2019-11-08:
Hi,

thank your for this cool project!
Currently I am setting up a TTGO-Beam and I can get PAX data on port1 and GPS data on port4, everything works fine. As I am trying to forward this data to a [mobile opensensemap station ](https://opensensemap.org/explore/5db5d4f89b884c001a14a334) i have to map everything to a single port.
[image]
Currently the station gets filled with sensordata, but obviously gps data is missing.
**how can I get GPS data on the same port?**

`#define GPSPORT                         4       // gps`
I tried setting GPSPORT to 1 as well but ended up with corrupt data
[image]
As you can see GPS data is sent on port1 but not in the way I expected

````
    if (port === 1) {
        // only wifi counter data, no gps
        if (bytes.length === 2) {
            return decode(bytes, [uint16], ['wifi']);
        }
        // wifi + ble counter data, no gps
        if (bytes.length === 4) {
            return decode(bytes, [uint16, uint16], ['wifi', 'ble']);
        }
        // combined wifi counter and gps data
        if (bytes.length === 15) {
            return decode(bytes, [uint16, latLng, latLng, uint8, hdop, uint16], ['wifi', 'latitude', 'longitude', 'sats', 'hdop', 'altitude']);
        }
        // combined wifi + ble counter and gps data
        if (bytes.length === 17) {
            return decode(bytes, [uint16, uint16, latLng, latLng, uint8, hdop, uint16], ['wifi', 'ble', 'latitude', 'longitude', 'sats', 'hdop', 'altitude']);
        }
    }
````
**what do I have to do to end up in the third case with combined pax/gps data?**

cyberman54, 2019-11-08:
You could solve this by modifying the payload decoder; thus i don't see a (breaking) change of paxcounter software?

nerdyscout, 2019-11-09:
well maybe I still do not understand enough of the paxcounter firmware, maybe I just don't  get your answer right... but my feeling is I have to modify the firmware, not the decoder.

As your [decoder](https://github.com/cyberman54/ESP32-Paxcounter/blob/master/src/TTN/plain_decoder.js) already has the option to receive PAX and GPS data on port1 I am asking myself how to configure the firmware to do so. Setting `#define GPSPORT 4 // gps` made both data been sent on the same port, but not within the same message. I was hoping there is an compile flag which will lead to send it simultaneously.

cyberman54, 2019-11-09:
Why not making a pull request for this?

cyberman54, 2019-11-11:
@nerdyscout I added a compile switch #define PAYLOAD_GPS in paxcounter.conf to [implement this function](https://github.com/cyberman54/ESP32-Paxcounter/commit/bc9d550b7da5585ba6d69c0e6b0a744dd3b244eb) in current development branch code. Could you please test if it works, and report here? Thanks.

nerdyscout, 2019-11-11:
hi @cyberman54 thank you for your support. from looking at your code I think this would do the job, I will try it some when the next few days. 
Currently I have my [own quick fix](https://github.com/ttnkn/ESP32-Paxcounter/tree/combine-lora-msg) running, this would allow to combine other messages as well. I will do the pull request when everything is in place and working.

cyberman54, 2019-11-12:
Took a quick look into your changes. Your approach looks good; that's the reason why i separated the payload.reset method.

cyberman54, 2019-11-17:
> hi @cyberman54 thank you for your support. from looking at your code I think this would do the job, I will try it some when the next few days.
> Currently I have my [own quick [...]

@nerdyscout any news here?

nerdyscout, 2019-11-17:
I simplified my first approach as this would have needed complex changes to the payload decoder.

The pr #489  currently works fine for me and seems not to break any existing code... but I spent now the whole day trying to figure out how to configure the [decoding options](https://docs.opensensemap.org/#api-Measurements-postNewMeasurement) of [opensensemap](https://opensensemap.org/).
https://forum.sensebox.de/t/mehrere-sensboxen-per-ttn-anbinden/725

I guess it is just some configuration issue, my code changes here should be fine. I just don't like to make another pr if I have to change further things to fix this.

cyberman54, 2019-11-18:
Nevertheless i already put this in development branch for testing. The port switch is a straight forward solution, better than compile switch - i like.

closed 2019-11-19 by cyberman54