cyberman54/ESP32-Paxcounter #491: PAX-Counter stoped sendig via LoRaWAN after first package

phisch92, 2019-11-17:
Dear all,

I am facing some issues with the actual release version 1.9.81.

My Hardware-Setup:
- [Adafruit HUZZAH32 - ESP32 Feather](https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/overview)
- [Radio FeatherWing for EU868](https://learn.adafruit.com/radio-featherwing/overview)
-[Adafruit OLED FeatherWing](https://learn.adafruit.com/adafruit-oled-featherwing)

As halfile I enabled octopus32.h.

This is my issue:
- After a reset the LoRa NWS receives the join request and succes to send the credentials
- After the send cycle (60s) passed I receive one pax-message on the TTN console
- Since my Test-Gateway is placed right next to the lora antenna I get a RSSI of -25 dBm and a SNR of 10 dB.
- After the inital package send via LoRaWAN I do not receive further messages
- Via console I receive the following messages:

`[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.81, runmode 0
[I][configmanager.cpp:190] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:55] open_storage(): Opening NVS
[I][configmanager.cpp:60] open_storage(): Done
[I][configmanager.cpp:204] loadConfig(): NVRAM settings version = 1.9.81
[I][configmanager.cpp:222] loadConfig(): bsecstate = 0
[I][configmanager.cpp:227] loadConfig(): loradr = 5
[I][configmanager.cpp:235] loadConfig(): txpower = 14
[I][configmanager.cpp:243] loadConfig(): adrmode = 1
[I][configmanager.cpp:251] loadConfig(): screensaver = 0
[I][configmanager.cpp:259] loadConfig(): screenon = 1
[I][configmanager.cpp:267] loadConfig(): countermode = 0
[I][configmanager.cpp:275] loadConfig(): sendcycle = 30
[I][configmanager.cpp:283] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:291] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:299] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:307] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:315] loadConfig(): blescantime = 8
[I][configmanager.cpp:323] loadConfig(): BLEscanmode = 1
[I][configmanager.cpp:331] loadConfig(): WIFIscanmode = 1
[I][configmanager.cpp:339] loadConfig(): rssilimit = 0
[I][configmanager.cpp:347] loadConfig(): payloadmask = 126
[I][configmanager.cpp:355] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:362] loadConfig(): Done
[I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:54] i2c_scan(): 0x3C: SSD1306 Display controller
[I][i2c.cpp:81] i2c_scan(): I2C scan done, 1 devices found.
[I][main.cpp:243] setup(): Starting LED Controller...
[I][main.cpp:279] setup(): Starting Bluetooth...
[I][blecsan.cpp:239] start_BLEscan(): Initializing bluetooth scanner ...
[I][blecsan.cpp:204] register_ble_callback(): Register GAP callback
[I][blecsan.cpp:228] register_ble_callback(): Set GAP scan parameters
[I][blecsan.cpp:252] start_BLEscan(): Bluetooth scanner started
[I][lorawan.cpp:303] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:306] lora_stack_init(): Starting LMIC...
[I][main.cpp:365] setup(): Starting Wifi...
[D][lorawan.cpp:501] myEventCallback(): JOINING
[I][main.cpp:378] setup(): Starting Interrupt Handler...
[I][bmesensor.cpp:58] bme_init(): BSEC v1.4.7.4
[E][bmesensor.cpp:135] checkIaqSensorStatus(): BME680 error -2
[E][bmesensor.cpp:65] bme_init(): BME680 sensor not found
[I][main.cpp:400] setup(): Starting Timers...
[I][main.cpp:465] setup(): Features: OLED LED BLE LORA FILTER PLAIN WIFI BME680
[D][lorawan.cpp:501] myEventCallback(): TXSTART
[D][lorawan.cpp:501] myEventCallback(): RXSTART
[I][lorawan.cpp:104] lora_setupForNetwork(): DEVaddr: 260124AE
[I][lorawan.cpp:108] lora_setupForNetwork(): Radio parameters: SF7 / BW125 / CR 4/5
[D][lorawan.cpp:501] myEventCallback(): JOINED
[D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1860 bytes left | Taskstate = 0
[D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2532 bytes left | Taskstate = 2
[D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 2580 bytes left | Taskstate = 2
[D][cyclic.cpp:61] doHousekeeping(): LEDloop 488 bytes left | Taskstate = 2
[... 177 lines cut ...]
[D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1428 bytes left | Taskstate = 2
[D][cyclic.cpp:61] doHousekeeping(): LEDloop 488 bytes left | Taskstate = 2
[I][cyclic.cpp:80] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[W][lorawan.cpp:363] lora_enqueuedata(): LORA sendqueue is full
[I][senddata.cpp:88] sendData(): Counter cleared
[W][lorawan.cpp:363] lora_enqueuedata(): LORA sendqueue is full
[D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1860 bytes left | Taskstate = 0
[D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2532 bytes left | Taskstate = 1
[D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1428 bytes left | Taskstate = 2
[D][cyclic.cpp:61] doHousekeeping(): LEDloop 488 bytes left | Taskstate = 2
[I][cyclic.cpp:80] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1860 bytes left | Taskstate = 0
[D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2532 bytes left | Taskstate = 1
[D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1428 bytes left | Taskstate = 2
[D][cyclic.cpp:61] doHousekeeping(): LEDloop 488 bytes left | Taskstate = 2
[I][cyclic.cpp:80] doHousekeeping(): BME680 Temp: 0.00°C | IAQ: 0.00 | IAQacc: 0
[W][lorawan.cpp:363] lora_enqueuedata(): LORA sendqueue is full
[I][senddata.cpp:88] sendData(): Counter cleared
[W][lorawan.cpp:363] lora_enqueuedata(): LORA sendqueue is full`

I think the major issue has something to do with the log:
'[W][lorawan.cpp:363] lora_enqueuedata(): LORA sendqueue is full'

I am thankful for any hints and help!
Thank you very much!

cyberman54, 2019-11-18:
The BME680 sensor on your board is not found, but configured in hal file.
Either your board has no BME680, or wrong i2c addess is selected, or you have a hardware issue.
Steps to solve:
1) disable BME680 sensor in hal file and re-test if LoRaWAN communication is working without BME680
2) enable BME680 and try to solve above issue with your sensor

Remember that you can't send many data over LoRaWAN network, since you will run in duty cycle limits, causing "sendqueue full" on the device. You need to carefully select sendcycle, datarate and volume of payload (e.g. number of sensors reported) to avoid this.

phisch92, 2019-11-18:
Thanky you very much for your quick respond.
I don't have any BME sensors attached to the MCU. Do I have to?

1.  I disabled HAS_BME, HAS_BME680 and BME680_ADDR in hal-file. Sadly the results did not change much:
- once the device restarts it succeds joining OTAA
- after the first send cycle passed I receive a package on the networkserver
- in order to receive further packages I have to restart the device.
- after a while I get the following waring:
> [W][lorawan.cpp:363] lora_enqueuedata(): LORA sendqueue is full

Here is a screenshot of the traffic
[image]

Sendcycle is set to 30 (=60s) in my setup. Since my gateway is not far I can send via SF7 and need about 50 ms airtime. I won't violate duty cycle regulations of 1%. However this is only for testing. I increased the Sendcycle to 120 (=240s) but I get the same issue. If I understand the issue correctly the lora buffer gets filled with PAX data but never sends it except for the first time. After a while the buffer is full.

> [I][reset.cpp:64] do_after_reset(): Starting Software v1.9.81, runmode 0

> [I][configmanager.cpp:190] loadConfig(): Reading settings from NVS
> [I][configmanager.cpp:55] open_storage(): Opening NVS
> [I][configmanager.cpp:60] open_storage(): Done
> [I][configmanager.cpp:204] [...]

Actually I set sendcycle in paxcounter.conf to 120. So I don't know why log says sendcycle = 30.

> [I][configmanager.cpp:283] loadConfig(): wifichancycle = 50
> [I][configmanager.cpp:291] loadConfig(): wifiantenna = 0
> [I][configmanager.cpp:299] loadConfig(): vendorfilter = 1
> [I][configmanager.cpp:307] loadConfig(): rgbluminosity = 30
> [I][configmanager.cpp:315] loadConfig(): blescantime = 8
> [I][configmanager.cpp:323] loadConfig(): BLEscanmode = 1
> [I][configmanager.cpp:331] loadConfig(): WIFIscanmode = 1
> [I][configmanager.cpp:339] loadConfig(): rssilimit = 0
> [I][configmanager.cpp:347] loadConfig(): payloadmask = 126
> [I][configmanager.cpp:355] loadConfig(): Monitor mode = 0
> [I][configmanager.cpp:362] loadConfig(): Done
> [I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan...
> [I][i2c.cpp:54] i2c_scan(): 0x3C: SSD1306 Display controller
> [I][i2c.cpp:81] i2c_scan(): I2C scan done, 1 devices found.
> [I][main.cpp:243] setup(): Starting LED Controller...
> [I][main.cpp:279] setup(): Starting Bluetooth...
> [I][blecsan.cpp:239] start_BLEscan(): Initializing bluetooth scanner ...
> [I][blecsan.cpp:204] register_ble_callback(): Register GAP callback
> [I][blecsan.cpp:228] register_ble_callback(): Set GAP scan parameters
> [I][blecsan.cpp:252] start_BLEscan(): Bluetooth scanner started
> [I][lorawan.cpp:303] lora_stack_init(): LORA send queue created, size 600 Bytes
> [I][lorawan.cpp:306] lora_stack_init(): Starting LMIC...
> [I][main.cpp:365] setup(): Starting Wifi...
> [D][lorawan.cpp:501] myEventCallback(): JOINING
> [I][main.cpp:378] setup(): Starting Interrupt Handler...
> [I][main.cpp:400] setup(): Starting Timers...
> [I][main.cpp:465] setup(): Features: OLED LED BLE LORA FILTER PLAIN WIFI BTN_PD
> [D][lorawan.cpp:501] myEventCallback(): TXSTART
> [D][lorawan.cpp:501] myEventCallback(): RXSTART
> [D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1864 bytes left | Taskstate = 0
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2604 bytes left | Taskstate = 1
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 2620 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 568 bytes left | Taskstate = 2
> [D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1864 bytes left | Taskstate = 0
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2604 bytes left | Taskstate = 1
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 2620 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 568 bytes left | Taskstate = 2
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:8896
[... 115 lines cut ...]
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2396 bytes left | Taskstate = 2
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1324 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 552 bytes left | Taskstate = 2
> [I][senddata.cpp:88] sendData(): Counter cleared

here it seems to miss out TXSTART, lorasend() and RXSTART

> [D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1864 bytes left | Taskstate = 0
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2396 bytes left | Taskstate = 2
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1324 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 552 bytes left | Taskstate = 2
> [D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1864 bytes left | Taskstate = 0
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2396 bytes left | Taskstate = 2
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1324 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 552 bytes left | Taskstate = 2
> [D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1864 bytes left | Taskstate = 0
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2396 bytes left | Taskstate = 2
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1324 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 552 bytes left | Taskstate = 2
> [D][cyclic.cpp:36] doHousekeeping(): IRQhandler 1864 bytes left | Taskstate = 0
> [D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2396 bytes left | Taskstate = 2
> [D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 1324 bytes left | Taskstate = 2       
> [D][cyclic.cpp:61] doHousekeeping(): LEDloop 552 bytes left | Taskstate = 2

Thank you for helping me!

cyberman54, 2019-11-19:
Settings in paxcounter.conf are used as factory defaults and don't override persistent runtime settings which are stored in NVRAM. To activate default settings, you have the options to send rcommand 0x09 during runtime r change the version number in platformio.ini, rebuild and reflash the code.

I can't reproduce the above issue with any of my test boards, but don't own a octopus, so i suspect it may be caused by the octopus settings.

cyberman54, 2019-11-19:
Make sure the LoRa Chip interrupt lines are both wired on your board to GPIOs of ESP32 cpu.

You need DIO0 and DIO1. In `octopus32.h` both lines use same GPIO33. That means on your board you need a suitable hardware which connects both, DIO0 and DIO1, with diodes to GPIO33. If your board hardware does not provide this circuit, LoRa send can't work properly with settings in `octopus32.h`, causing your sendqueue runs full.

cyberman54, 2019-11-19:
@phisch92 Did you follow the [instructions here](https://github.com/BCX18ConnectedLife/octopus-lorawan/blob/master/ASSEMBLY.md) and placed the 2 diodes on the Radio Feather Wing board?

phisch92, 2019-11-19:
Perfect - that is the solution. I connected DIO1 to its own pin (27) and adjusted the hal file accordingly. I will play around with the hardware setup and once everything works I will upload a new hal file for the board.
Thank you very much!

closed 2019-11-19 by cyberman54