cyberman54/ESP32-Paxcounter #498: GPS Data

digitalyours, 2019-11-27:
I'm struggling with getting GPS data transmitted from my TTGO T-Beam (Start-up sequence [1]). I'm only seeing data over port 01, nothing else can be seen in the TTN monitor. Also I don't see any data transmitted to TTN Mapper integration. This is maybe not a bug, since I'm new to the whole topic. Am I overlooking something?

Best & thx
Christoph

[1]
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.81, runmode 0
[I][power.cpp:119] AXP192_init(): AXP192 PMU initialization failed
[I][configmanager.cpp:190] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:55] open_storage(): Opening NVS
[I][configmanager.cpp:60] open_storage(): Done
[I][configmanager.cpp:204] loadConfig(): NVRAM settings version = 1.9.81
[I][configmanager.cpp:222] loadConfig(): bsecstate = 0
[I][configmanager.cpp:227] loadConfig(): loradr = 5
[I][configmanager.cpp:235] loadConfig(): txpower = 14
[I][configmanager.cpp:243] loadConfig(): adrmode = 1
[I][configmanager.cpp:251] loadConfig(): screensaver = 0
[I][configmanager.cpp:259] loadConfig(): screenon = 1
[I][configmanager.cpp:267] loadConfig(): countermode = 0
[I][configmanager.cpp:275] loadConfig(): sendcycle = 30
[I][configmanager.cpp:283] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:291] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:299] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:307] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:315] loadConfig(): blescantime = 8
[I][configmanager.cpp:323] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:331] loadConfig(): WIFIscanmode = 1
[I][configmanager.cpp:339] loadConfig(): rssilimit = 0
[I][configmanager.cpp:347] loadConfig(): payloadmask = 126
[I][configmanager.cpp:355] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:362] loadConfig(): Done
[I][i2c.cpp:33] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:81] i2c_scan(): I2C scan done, 0 devices found.
[I][main.cpp:209] setup(): PSRAM found and initialized
[I][gpsread.cpp:37] gps_init(): Using serial GPS
[I][main.cpp:295] setup(): Starting GPS Feed...
[I][lorawan.cpp:303] lora_stack_init(): LORA send queue created, size 600 Bytes
[I][lorawan.cpp:306] lora_stack_init(): Starting LMIC...
[I][main.cpp:365] setup(): Starting Wifi...
[D][lorawan.cpp:501] myEventCallback(): JOINING         
[I][main.cpp:378] setup(): Starting Interrupt Handler...
[I][main.cpp:400] setup(): Starting Timers...
[I][main.cpp:458] setup(): Starting Timekeeper...
[I][timekeeper.cpp:124] timepulse_init(): Timepulse: external (GPS)
[I][main.cpp:465] setup(): Features: PMU OLED PSRAM BATT BLE GPS LORA FILTER PACKED WIFI BTN_PD

[D][cyclic.cpp:36] doHousekeeping(): IRQhandler 2376 bytes left | Taskstate = 0
[D][cyclic.cpp:39] doHousekeeping(): LMiCtask 2584 bytes left | Taskstate = 2
[D][cyclic.cpp:42] doHousekeeping(): Lorasendtask 2616 bytes left | Taskstate = 2
[D][cyclic.cpp:46] doHousekeeping(): Gpsloop 1592 bytes left | Taskstate = 1
[I][cyclic.cpp:68] doHousekeeping(): Battery: external power
[I][power.cpp:102] AXP192_showstatus(): Battery charging, -0.00V @ -3mAh
[I][power.cpp:110] AXP192_showstatus(): USB powered, 0mW

cyberman54, 2019-11-27:
Did you study the hints in readme.md how to configure the settings to generate combined messages (pax and gps) on port 1, suitable for ttnmapper?

cyberman54, 2019-11-27:
"GPS data can also be combined with payload on port 1, #define GPSPORT 1 in paxcounter.conf to enable"

digitalyours, 2019-11-27:
Yes, I did and just double checked. No GPS data gets transmitted on port 1, just PAX and WIFI...

cyberman54, 2019-11-27:
@digitalyours sure that you GPS has a fix? If not, intentionally no data is transmitted.
Indicator of "has a fix" is that red PPS LED on T-BEAM is flashing.
Please confirm, thanks.

digitalyours, 2019-11-27:
Ok, thanks for pointing this out. Indeed there is no red PPS LED flashing. Doesn't matter where I place the board, maybe the GPS is broken. :(

cyberman54, 2019-11-27:
Paxcounter software does not (yet) configure the GPS chip. Thus the chip will run in default configuration as delivered by your hardware source. If this configuration is faulty, this may cause fix issues.

nerdyscout, 2019-11-27:
actually this is the reason why #489 is still on draft. Although my device got the fixled blinking, I can the in the terminal `sendData(): No valid GPS position`

cyberman54, 2019-11-27:
solved by @nerdyscout PR, thanks!

closed 2019-11-27 by cyberman54

digitalyours, 2019-11-27:
Many thx for the pr. It worked for me!