cyberman54/ESP32-Paxcounter #503: LoRaWAN: in case of .... where to look in the code?

AugustQu, 2019-12-05:
Hi,

my PaxCounter works, hardware and software (including the SD-card-sw) but the LoRaWAN-connection does not work.

There are LoRaWAN-gateways in a city nearby so I tried it over there. I can report:
- in TTN in the console I can see: Status (green) 3 days ago
- but I do not see any data
- on the board: a red led (on), a blue led (on) and a green led (flashing)

Where in the code can I look to see what's going wrong?

I am able to add some log-statements into the code.

Any tips please.

AugustQ

cyberman54, 2019-12-05:
use LMIC debug mode, can be switched in lmic_config.h and use app debug level 4 in platformio.ini

AugustQu, 2019-12-06:
OK, will try it tomorrow.

AugustQu, 2019-12-25:
Hi,

problem is solved.

For some weeks we have a LoRaWAN-gateway here in the area so I could test more often. I found 2 problems:
- I need to wait a bit longer
- I need to go outside of the house.

Fault was on my side.

closed 2019-12-25 by cyberman54