cyberman54/ESP32-Paxcounter #505: Version 1.9.8 on LoLin32lite board + LoraNode32-Lite shield with OLED display (SSD1306)

aemmenet, 2019-12-09:
I’m running Paxcounter version 1.7.41 on two stacked LoLin32lite board + LoraNode32-Lite shield with a OLED display (SSD1306) on the I2C bus showing the detailed status. After updating the first one to Paxcounter version 1.9.8 the OLED display stays dark. LoRa & SPI are fine. The BitBang_I2C sample sketch shows “Device found at 0x3C, type = SSD1306” and the SS_OLED sample sketch shows the animation. But Paxcounter says “I2C scan done, 0 devices found”.

// Pins for I2C interface of OLED Display.
#define MY_OLED_SDA (14)
#define MY_OLED_SCL (12)
#define MY_OLED_RST NOT_A_PIN

Any debug hint?

cyberman54, 2019-12-10:
LoLin32lite board has no OLED display. Please add detailed information on what you build. Check your wirings.

cyberman54, 2019-12-30:
Stale.

closed 2019-12-30 by cyberman54

aemmenet, 2019-12-31:
The LoLin32lite board + LoraNode32-Lite shield with a OLED display (SSD1306) on the I2C bus is working fine till Version 1.8.2. After changing to ss_oled / BitBang_I2C the display is no longer found.

FYI: lolin32litelora.h has wrong pin definitions for LoRa! Check https://github.com/hallard/ESP32-Paxcounter/blob/master/src/hal/lolin32litelora.h for the right pins.

aemmenet, 2019-12-31:
This pin definitions are working fine. Only the display is missing after 1.8.2.

// https://github.com/hallard/ESP32-Paxcounter/blob/master/src/hal/lolin32litelora.h

// Pins for LORA chip SPI interface, reset line and interrupt lines
#define LORA_SCK  (18)	// was Pin 5 
#define LORA_CS   (5)   // was Pin 18, also named SS
#define LORA_MISO (19)
#define LORA_MOSI (23)	// was Pin 27
#define LORA_RST  (25)
#define LORA_IRQ  (27)  // also named DIO0
#define LORA_IO1  (26)  // also named DIO1
#define LORA_IO2  LMIC_UNUSED_PIN   // Pin 4
#define LORA_IO5  LMIC_UNUSED_PIN   // Pin 35

aemmenet, 2019-12-31:
The hardware is fine because the BitBang_I2C sample sketch shows “Device found at 0x3C, type = SSD1306” and the SS_OLED sample sketch shows the animation. So there must still be be a issue with the new OLED driver in Paxcounter.

aemmenet, 2020-01-01:
According to the README.md file the WeMos LoLin32 + LoraNode32 shield is a Paxcounter supported ESP32 based board. The LoRa shield is from Charles Hallard. As there ist no firmware available for this board he recommends Paxcounter for testing. Charles Hallard ist also mentionend as a major code contributor for the Paxcounter project. So he may help fixing the display problem with his shield.

cyberman54, 2020-01-01:
@aemmenet this is an open source project. Why not trying to fix by yourself and make a PR?

cyberman54, 2020-01-01:
@aemmenet i found that i2c bus lines were swapped in the hal file. I pushed an update recently, please test, thanks.

aemmenet, 2020-01-01:
The LoRa pins are ok now. The original i2c pins were already ok according to Charles Hallard's hal-file. But even with exchanged pins, the display remains dark. I'm trying to analyze the new "after version 1.8.2" i2c communication by inserting additional debug code. I can use the working sample sketches for BitBang_I2C and ss_oled as a guide. 
I have to get to know github first to be able to do my own PR! Due to a lack of programming knowledge, my debugging still works according to the "try and error" principle.
It seems that I'm the first after Charles Hallard running Paxcounter on this board with stacked LoRa-shield.