cyberman54/ESP32-Paxcounter #508: Blurry Display on HeltecV2

tomschlde, 2019-12-23:
After Updating to the current version the display becomes blurry on hektec V2 - in the former versions the content was excellent. Seems to try grayscale - see screenshot.
[image]

cyberman54, 2019-12-23:
Could you please try to increase display contrast value in paxcounter
conf and retry?

tomschlde, 2019-12-23:
Got it... burn in - the text in the current version is not at the same coodinates.
Can be seen on the other screens.
Have to buy a new one :)

[image]

closed 2019-12-23 by tomschlde

cyberman54, 2019-12-23:
That's right, the mask was changed, and OLEDs wear out pretty fast. That's why in the new version the contrast was decreased.