cyberman54/ESP32-Paxcounter #510: combined GPS+COUNTERPORT without BLECOUNTER gives JSON Error in Converter

7samurai, 2019-12-27:
Hi,
i ran into this issue, and get an Error("json: unsupported value: NaN") in the Payload Console. I Use the standard payload / converter
My Payload is 15 bytes long, so 
```
if (bytes.length === 4 || bytes.length > 15) {
      decoded.ble = (bytes[i++] << 8) | bytes[i++];
    }
```
has no effect. In the converter there is on row 9
```
converted.pax = converted.ble + converted.wifi;
```
what cause the JSON Error. For me, its simple to remove converted.ble, but if some other ran into this issus... Maybe add an check if value ble is numeric oder wifi?

cyberman54, 2019-12-27:
Thanks for reporting this bug. Could you please try this quick & dirty hack:

add two lines
```
 decoded.wifi = 0;
 decoded.ble = 0;
```
in `plain_decoder.js`

```
function Decoder(bytes, port) {
  var decoded = {};
  decoded.wifi = 0;
  decoded.ble = 0;

```

cyberman54, 2019-12-30:
fixed.

closed 2019-12-30 by cyberman54