cyberman54/ESP32-Paxcounter #521: Refactor gps decoding using sparkfun lib

cyberman54, 2020-01-04:


hansju-11, 2021-04-11:
Maybe we can use preprocessor to define something like GPS_LIB_TIP  1 or 2.
1 or not set means use TinyGSP and 2 means use Ublox. 
This will fix #671 too.

Second, can somebody test, if the ublox can parse NMEA over serial an i2c from other devices? I do not have an Quectel right now.

cyberman54, 2021-04-11:
What about any license restrictions on ublox lib? Paxcounter is open source.

hansju-11, 2021-04-12:
It is MIT License. 

From my point of view it should be ok, as long if we include a remark on that license.

 

 

https://github.com/sparkfun/SparkFun_u-Blox_NEO_M9N/blob/master/LICENSE.md

 

 

 

Gesendet: Sonntag, 11. April 2021 um 21:01 Uhr

cyberman54, 2021-04-12:
I think it could with generic NMEA GNSS outputs, too. Look in this example:
https://github.com/sparkfun/SparkFun_u-blox_GNSS_Arduino_Library/blob/master/examples/Example2_NMEAParsing/Example2_NMEAParsing.ino

hansju-11, 2021-05-03:
Short Update, 
I did some test with the LIB, but I did not get it to work. 
It looks like per default the NEO6 is sendig only NMEA infos. If I set the output to UBLOX only I did not get any infos. I can still send commands in ublox. It looks like there is an output filter active. Than I set up a path through from GPS to usb/Serial and use the u-center software. Here I can change the config and setup the right outputfilter and I got ublox infos. 
But when I switch bach to the lib I got only wrong data.
I'm still in hope :-) maybut somebody has some other information and can share this here. I will try to see if we get this to run.

cyberman54, 2021-05-07:
@hansju-11 another option could be the [NeoGPS](https://github.com/SlashDevin/NeoGPS) library.

cyberman54, 2022-02-16:
Solved in v3.2.0 without sparkfun lib, but by own code :-)

closed 2022-02-16 by cyberman54

hansju-11, 2022-03-10:
> Solved in v3.2.0 without sparkfun lib, but by own code :-)

Cool