cyberman54/ESP32-Paxcounter #523: Screencycling

rdmueller, 2020-01-19:
for devices without a button, it would be nice if the screen/display cycles every x seconds.
In addition, it would be great if this mode skips screens without information (no GPS, no BME).

cyberman54, 2020-01-19:
Good idea, let's see if anyone implements it.

rdmueller, 2020-01-19:
I am new to the paxcounter, but I guess I am already halfway through the implementation...

cyberman54, 2020-01-19:
Page skip is ready (experimental), look [here](https://github.com/cyberman54/ESP32-Paxcounter/commit/e8abe8567bca1d618a4357c2e4163f8db40535b5)

Page cycling yet to come.

cyberman54, 2020-01-20:
@rdmueller Done:
https://github.com/cyberman54/ESP32-Paxcounter/commit/f960627910be691656addaf713e03b5237a20aee
Could you please test and report here? Thanks.

cyberman54, 2020-01-25:
Done.

closed 2020-01-25 by cyberman54

rdmueller, 2020-01-25:
sorry, it took a while but today I found some time to give it a try. Works like a charm!