cyberman54/ESP32-Paxcounter #537: Add support for M5 stack boards

samuk, 2020-01-25:
Would it be possible to add support for this board
https://m5stack.com/collections/m5-core/products/basic-core-iot-development-kit

With this shield
https://m5stack.com/products/lora-module-868mhz

And GPS
https://m5stack.com/collections/m5-module/products/gps-module

cyberman54, 2020-01-25:
M5 Lora module has a Lora modem (RHF76-052) with it's own firmware and AT command set. This is not compatible with paxcounter code, requiring direct access to LoRa transceiver chip (SX127x). So unfortunately i will close this issue without a solution.

samuk, 2020-01-25:
Ah that's a shame. I assume this RHF76-052 module is the same?

https://m5stack.com/collections/m5-module/products/lorawan-modulerhf76-052

samuk, 2020-01-25:
That's a shame.

Is it true of the  RA018 as well?
https://www.banggood.com/M5Stack-Wireless-LoRa-Module-868MHz-Communicate-Module-Ra-01H-with-Prototyping-Area-SPI-Communication-Protoco-p-1600755.htm

cyberman54, 2020-01-25:
@samuk RA01 would work, it has a native SX1278 chip.

samuk, 2020-01-25:
The chip seems to be marked 803MHZ - 930MHZ? 

Eg https://www.aliexpress.com/item/4000505879977.html

cyberman54, 2020-01-25:
That one has a SX1276 for 868 MHz, yes.

cyberman54, 2020-01-26:
@samuk if you grab all pin mappings properly, i can make a version for M5. Can you test it then?

samuk, 2020-01-26:
Sure, I don't have the board yet, but I can order today and test when it arrives.

Looks like the core module pinout is [here](https://docs.m5stack.com/#/en/core/gray?id=peripherals-pin-map)

cyberman54, 2020-01-26:
The above schematics is for LORA chip RHF76-052, which does not expose the SPI interface of SC127x, so is not compatible to paxcounter code. We need pin mappings for LORA board with RA01 chip.

cyberman54, 2020-01-26:
@samuk i checked the pinmappings, it should work with settings given below. 
On the LORA pcb there will be 1 wire to be soldered to get the board running with LMIC.
The LCD display should run with [this library](https://github.com/bitbank2/bb_spi_lcd), which has similar commands as the OLED library by same author. Thus, shouldn't be to much effort to get paxcounter running on M5stack hardware. Good luck!

```
// clang-format off
// upload_speed 921600
// board M5Stack-Core-ESP32

#ifndef _M5STACK_H
#define _M5STACK_H

#include <stdint.h>

#define HAS_LORA 1 // comment out if device shall not send data via LoRa or has no M5 RA01 LoRa module

// Pins for LORA chip SPI interface, reset line and interrupt lines
#define LORA_SCK  SCK
#define LORA_CS   SS
#define LORA_MISO MISO
#define LORA_MOSI MOSI
#define LORA_RST  GPIO_NUM_36
#define LORA_IRQ  GPIO_NUM_26
#define LORA_IO1  GPIO_NUM_34 // must be externally wired on PCB!
#define LORA_IO2  LMIC_UNUSED_PIN

// enable only if you want to store a local paxcount table on the device
#define HAS_SDCARD  1      // this board has an SD-card-reader/writer
// Pins for SD-card
#define SDCARD_CS    GPIO_NUM_4
#define SDCARD_MOSI  MOSI
#define SDCARD_MISO  MISC
#define SDCARD_SCLK  SCK

// user defined sensors
//#define HAS_SENSORS 1 // comment out if device has user defined sensors

#define CFG_sx1276_radio 1 // select LoRa chip
#define BOARD_HAS_PSRAM // use if board has external PSRAM
#define DISABLE_BROWNOUT 1 // comment out if you want to keep brownout feature

//#define HAS_DISPLAY 1
//#define DISPLAY_FLIP  1 // use if display is rotated
//#define BAT_MEASURE_ADC ADC1_GPIO35_CHANNEL // battery probe GPIO pin -> ADC1_CHANNEL_7
//#define BAT_VOLTAGE_DIVIDER 2 // voltage divider 100k/100k on board

//#define HAS_LED (21) // on board  LED
#define HAS_BUTTON (39) // on board button A

// GPS settings
#define HAS_GPS 1 // use on board GPS
#define GPS_SERIAL 9600, SERIAL_8N1, RXD2, TXD2 // UBlox NEO 6M RX, TX
#define GPS_INT GPIO_NUM_35 // 30ns accurary timepulse, to be external wired on pcb: shorten R12!

// Pins for interface of LC Display
#define MY_OLED_CS GPIO_NUM_14
#define MY_OLED_DC GPIO_NUM_27
#define MY_OLED_CLK GPIO_NUM_18
#define MY_OLED_RST GPIO_NUM_33
#define MY_OLED_BL GPIO_NUM_32
#define MY_OLED_MOSI GPIO_NUM_23
#define MY_OLED_MISO GPIO_NUM_23

#endif
```

samuk, 2020-01-27:
Thanks! I've ordered the board and will test when it arrives and report back.

cyberman54, 2020-01-27:
#define LORA_IO1  GPIO_NUM_34 // must be externally wired on PCB!

That means you must solder a short wire on the LORA 868 pcb, from DIO1 (Lora RA01 chip) to GPIO34 (ESP32 CPU bus).

cyberman54, 2020-01-27:
Experimental version is ready to test:
https://github.com/cyberman54/ESP32-Paxcounter/commit/edf038d6f2adc1ac637e7641bd3bc7ff209cf27c

chopmann, 2020-02-03:
@cyberman54 I got the M5 Stack "Fire" and will test the code. Don't have a LoRa Module, but might be able to help with the display.

samuk, 2020-02-03:
Mine is delayed due to Coronavirus but will report back once I get it.

cyberman54, 2020-02-04:
> @cyberman54 I got the M5 Stack "Fire" and will test the code. Don't have a LoRa Module

for this, unset "HAS_LORA" by commenting out in m5stack.h file.

chopmann, 2020-02-05:
@cyberman54 
The board should be called:
```
[env:m5stack-core-esp32]  
platform = espressif32
board = m5stack-core-esp32
```
After that it compiles. Should the display be done in `display.cpp` with ifdef and so on or would you prefer another way?

cyberman54, 2020-02-05:
@chopmann thanks for the hint, please change the board name in the hal file. I tested on a windows system only; probably it's case sensitive on linux system.

Yes, plan is to have lcd support in display.cpp with #ifdef.

cyberman54, 2020-02-08:
Experimental M5 support is now integrated in master.

closed 2020-02-08 by cyberman54

TheHysiK, 2020-04-29:
Hi! Any news on M5Stack with LoRa Module? Did anyone test it?
I would like to order two of them for trying Paxcounter but I see it is experimental, will it work like TTGo model? I plan on using 1 module for WIFI and the other for BLE as individual scanners, is that better isn't i?t

cyberman54, 2020-04-30:
M5 Lora Board support is already integrated in main branch code, and works. I tested it. But it is necessary to add one wire on the LoRa board by hand. See hal file for further details.

TheHysiK, 2020-04-30:
> M5 Lora Board support is already integrated in main branch code, and works. I tested it. But it is necessary to add one wire on the LoRa board by hand. See hal file for further details.
I have read this: "from DIO1 (Lora RA01 chip) to GPIO34 (ESP32 CPU bus)." it os ok, I just wanted to know that it really worked. Ordering 2 units with LoRa module, will update when I receive them, thanks again and regards!

TheHysiK, 2020-05-12:
@cyberman54 Hi!
I got my m5stack core board, soldered cable and uploaded paxcounter.
When I start module and monitor via USB it starts with QR code but it enters in a reboot cycle. Terminal data is:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[W][esp32-hal-psram.c:30] psramInit(): PSRAM init failed!
[I][reset.cpp:64] do_after_reset(): Starting Software v1.9.991, runmode 0
[I][configmanager.cpp:191] loadConfig(): Reading settings from NVS
[I][configmanager.cpp:56] open_storage(): Opening NVS
[I][configmanager.cpp:61] open_storage(): Done
[I][configmanager.cpp:205] loadConfig(): NVRAM settings version = 1.9.991
[I][configmanager.cpp:223] loadConfig(): bsecstate = 0
[I][configmanager.cpp:228] loadConfig(): loradr = 5
[I][configmanager.cpp:236] loadConfig(): txpower = 14
[I][configmanager.cpp:244] loadConfig(): adrmode = 1
[I][configmanager.cpp:252] loadConfig(): screensaver = 0
[I][configmanager.cpp:260] loadConfig(): screenon = 1
[I][configmanager.cpp:268] loadConfig(): countermode = 0
[I][configmanager.cpp:276] loadConfig(): sendcycle = 30
[I][configmanager.cpp:284] loadConfig(): wifichancycle = 50
[I][configmanager.cpp:292] loadConfig(): wifiantenna = 0
[I][configmanager.cpp:300] loadConfig(): vendorfilter = 1
[I][configmanager.cpp:308] loadConfig(): rgbluminosity = 30
[I][configmanager.cpp:316] loadConfig(): blescantime = 8
[I][configmanager.cpp:324] loadConfig(): BLEscanmode = 0
[I][configmanager.cpp:332] loadConfig(): WIFIscanmode = 1
[I][configmanager.cpp:340] loadConfig(): rssilimit = 0
[I][configmanager.cpp:348] loadConfig(): payloadmask = 127
[I][configmanager.cpp:356] loadConfig(): Monitor mode = 0
[I][configmanager.cpp:363] loadConfig(): Done
[I][i2c.cpp:22] i2c_scan(): Starting I2C bus scan...
[I][i2c.cpp:56] i2c_scan(): 0x75: IP5306 power management
[I][i2c.cpp:74] i2c_scan(): I2C scan done, 1 devices found.
assertion "psramFound()" failed: file "src\main.cpp", line 208, function: void setup()
abort() was called at PC 0x40137033 on core 1

Backtrace: 0x40094df8:0x3ffd2000 0x40095029:0x3ffd2020 0x40137033:0x3ffd2040 0x400d53f3:0x3ffd2070 0x400ea423:0x3ffd2110 0x40090f55:0x3ffd2130

cyberman54, 2020-05-15:
Thanks for this hint.
I found, with a (newer?) M5fire PSRAM works.
Maybe M5core does not have the additional PSRAM chip.

TheHysiK, 2020-05-17:
> Thanks for this hint.
> I found, with a (newer?) M5fire PSRAM works.
> Maybe M5core does not have the additional PSRAM chip.

Yes, M5core does not have PSRAM chip checked specs from webpage.
For now module is working perfectly with that.

FrankUlbrich, 2020-06-10:
https://www.bjoerns-techblog.de/2019/04/ttn-network-tester/

In diesem Artikel gibt es ebenfalls Anmerkungen zu den 2 Schields und M5 Fire bezüglich der Pins.

bCyberBasti, 2020-10-11:
[image]

Before i solder the wrong pins i want to double check 😉  are my marked solder points right?

cyberman54, 2020-10-11:
Yes, looks good.

[image]

revkillj0y, 2022-04-13:
Has anyone had gotten this module working on a Core2?

cyberman54, 2022-04-13:
@revkillj0y i checked the pin mappings of core2. Power of some peripherals is controlled by AXP192 PMU chip, which means that these pins need to be properly initialized by paxcounter. No big deal, since paxcounter already has AXP192 support for T-Beam, but requires some enhancements in `power.cpp`. 

Moreover, core2 uses 2 SPI buses (GPIO 21/22 & GPIO 33/35). Again, no big deal, logic could be copied from paxcounter setup for Heltec V2 board, which uses 2 SPI buses, too.

Maybe your chance to create a PR for this?

[image]